RobotLocomotion/drake #2139: Race Condition Between Generating and Using LCM Messages

liangfok, 2016-04-21:
In [CI build 962, gcc/linux-matlab](https://drake-jenkins.csail.mit.edu/job/experimental/962/compiler=gcc,label=linux_matlab/consoleText), which is part of PR https://github.com/RobotLocomotion/drake/pull/2103, the following error occurred on line 27776:

```
27776 /home/ubuntu/workspace/experimental/06107e1d/drake/examples/Cars/publishDrivingCommand.cpp:5:57: fatal error: lcmtypes/drake/lcmt_driving_control_cmd_t.hpp: No such file or directory
27777  #include "lcmtypes/drake/lcmt_driving_control_cmd_t.hpp"

```
The problem is `lcmt_driving_control_cmt_t.hpp` is not generated on line 30628:

```
30628 [ 25%] Generating lcmgen/lcmtypes/drake/lcmt_driving_control_cmd_t.hpp
```

Thus, I believe there is a missing dependency specification that ensures LCM messages are generated before consumers of the generated message header files are compiled.

rpoyner-tri, 2016-04-21:
something like this maybe?
`
  add_dependencies(<YOUR_TARGET>_lcmtypes lcmtype_agg_hpp)
`

liangfok, 2016-04-21:
Thanks @rpoyner-tri.

The auto-generated `lcmt_driving_control_cmd_t.hpp` is used in both of the following files:
* drake/examples/Cars/simulateLCM.cpp
* drake/examples/Cars/publishDrivingCommand.cpp

In drake/examples/Cars/CMakeLists.txt, they are compiled into the following targets, respectively: `carSimLCM`, and `publishDrivingCommand`.

Looking in drake/CMakeLists.txt, I believe the LCM message targets are defined in drake/cmake/lcmtypes.cmake. It is in this file that I see the `lcmtype_agg_hpp` CMake target:

```
    # add a custom target to force generation of aggregate hpp
    add_custom_target(lcmtype_agg_hpp ALL
      DEPENDS ${LCMTYPES_CPP_HEADERFILES})
```

Looking through the other CMakeLists.txt in Drake, I also see a `drake_lcmtypes` dependency being added alongside `lcmtype_agg_hpp`. I'm not sure where that target is being defined, but I'll go ahead and throw it in to be consistent.

billhoffman, 2016-04-21:
I see the same error here:
https://drake-jenkins.csail.mit.edu/job/experimental/965/compiler=gcc,label=linux_matlab/consoleFull

billhoffman, 2016-04-21:
publishDrivingCommand and carSimLCM need to have:  
add_dependencies(runQuadrotorDynamics drake_lcmtypes lcmtype_agg_hpp)

You can see the same thing being done for other examples:
examples/Quadrotor/CMakeLists.txt:  add_dependencies(runQuadrotorDynamics drake_lcmtypes lcmtype_agg_hpp)
examples/Quadrotor/CMakeLists.txt:  add_dependencies(runQuadrotorLQR drake_lcmtypes lcmtype_agg_hpp)
examples/SimpleCar/CMakeLists.txt:  add_dependencies(drakeSimpleCar drake_lcmtypes lcmtype_agg_hpp)

liangfok, 2016-04-21:
@billhoffman I will update this PR with the fix. Thanks for catching it!

liangfok, 2016-04-21:
@billhoffman, sorry about the confusing previous message. I already submitted a PR here: https://github.com/RobotLocomotion/drake/pull/2141

I'll assign it to you for a quick look.

billhoffman, 2016-04-21:
Yes, I see it and it looks good.  Should fix the random matlab test failures that I thought were my fault from the conversion to cmakes add_test for matlab somehow.  :)

closed 2016-04-21 by jwnimmer-tri