RobotLocomotion/drake #2147: LINK : warning LNK4044: unrecognized option '/pthread'; ignored

billhoffman, 2016-04-21:
https://github.com/RobotLocomotion/drake/blob/master/drake/CMakeLists.txt#L190 unconditionally adds -pthread with no checks on the compiler or platform being used.  Came from this commit:  https://github.com/RobotLocomotion/drake/commit/9f730dad353389947ed3110f3f392285c8bf144e, looks like from @david-german-tri .

sherm1, 2016-04-21:
:+1: for fixing this -- it's annoying on Windows.

billhoffman, 2016-05-06:
There is also a fPIC showing up on windows.   People should not add compiler specific flags into the cmake files without guarding them with checks on the compiler being used.  There are several places where -fPIC is being added without checking the system as well.

mwoehlke-kitware, 2016-05-10:
Is the `-pthread` is even needed? I removed it, and things are still building...?

Also, (@david-german-tri) what is the purpose of all the `GTEST_FOUND` checks when gtest is `REQUIRED`?

david-german-tri, 2016-05-10:
I vaguely recall that `-pthread` was a hack to fix the Mac build.  If I'm wrong and it's unnecessary, then please feel free to delete it!

I started adding `GTEST_FOUND` checks to hedge my bets, since there was some pushback on making gtest `REQUIRED`.  No one has complained about that in a while, so perhaps we can drop them.

mwoehlke-kitware, 2016-05-10:
> I vaguely recall that `-pthread` was a hack to fix the Mac build. If I'm wrong and it's unnecessary, then please feel free to delete it!

I'll remove it and check the CI once there is a PR. If it breaks, I'll add a proper (or at least platform-specific) fix.

> I started adding `GTEST_FOUND` checks to hedge my bets, since there was some pushback on making gtest `REQUIRED`.  No one has complained about that in a while, so perhaps we can drop them.

I probably won't be so ambitious as to systematically remove them :smile:; just wanted to make sure I wasn't missing something.

Thanks.

PR #2313 (2016-05-12): Portable flags
Remove some platform-specific flags that were being set without platform checks, and are also superseded by different, "preferred" mechanisms to accomplish the same goals in modern CMake.

Fixes #2147.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/2313)
closed 2016-05-12 by david-german-tri