RobotLocomotion/drake #2152: RigidBodySystem Needs To Provide More Info About Semantics Of Output Signals

liangfok, 2016-04-21:
The output signal of `RigidBodySystem` is defined as follows (see [RigidbodySystem:141](https://github.com/RobotLocomotion/drake/blob/5878f08970311a3b24ea58fe32bfa8be1f97225b/drake/systems/plants/RigidBodySystem.h#L141)):

```
  template <typename ScalarType>
  using OutputVector = Eigen::Matrix<ScalarType, Eigen::Dynamic, 1>;
```
In the context of the the car simulation, `ScalarType` is a `double`. Unfortunately, a vector of doubles does not provide any information about the meaning of the output values. 

The first part of the output vector consists of the positions and velocities of the joints in the `RigidBodyTree` that's contained within the `RigidBodySystem`. This information is stored in the `RigidBody` objects within the `RigidBodyTree` and is initialized in `RigidBodyTree::compile()`. The relevant code starts on [RigidBodyTree:131](https://github.com/RobotLocomotion/drake/blob/32d83aab069a8cd262beabd52efd8a177c2981e4/drake/systems/plants/RigidBodyTree.cpp#L131):

```
  num_positions = 0;
  num_velocities = 0;
  for (auto it = bodies.begin(); it != bodies.end(); ++it) {
    RigidBody& body = **it;
    if (body.hasParent()) {
      body.position_num_start = num_positions;
      num_positions += body.getJoint().getNumPositions();
      body.velocity_num_start = num_velocities;
      num_velocities += body.getJoint().getNumVelocities();
    } else {
      body.position_num_start = 0;
      body.velocity_num_start = 0;
    }
  }
```
Using the variables `RigidBodyTree::num_positions`, `RigidBodyTree::num_velocities`, `RigidBody::position_num_start`, the current position and velocities of each DOF can be extracted from the output vector of `RigidBodySystem`.

The remaining part of `RigidBodySystem`'s output vector consists of sensor states. Unfortunately, it is currently not possible to determine the semantics of this data. There's a [System Framework 2.0 design](https://docs.google.com/document/d/1YQIh2LUETE4CrQ6NvVeNefiIgXch5Q9JO0OWPyMOe90/edit) that should eventually resolve this problem, but we need a short term solution in the mean time. I created this issue to track the changes necessary to a consumer of RigidBodySystem's output can extract the sensor state contained within this output.

liangfok, 2016-04-22:
The sensor data portion of `RigidBodySystem`'s output vector  is saved by the `for` loop starting at [RigidBodySystem.cpp#L221](https://github.com/RobotLocomotion/drake/blob/496eaf74b869835f44874d7fb31976a9399490a9/drake/systems/plants/RigidBodySystem.cpp#L221). I've replicated it below.

```
  int index = getNumStates();
  for (const auto& s : sensors) {
    y.segment(index, s->getNumOutputs()) = s->output(t, kinsol, u);
    index += s->getNumOutputs();
  }
```

As shown by the code above, the output signals of the sensors are stored serially in the order they appear in [RigidBodySystems::sensors](https://github.com/RobotLocomotion/drake/blob/496eaf74b869835f44874d7fb31976a9399490a9/drake/systems/plants/RigidBodySystem.h#L247).

liangfok, 2016-04-22:
Given the above analysis, for the sake of time, and considering the fact that we are talking about Drake's System Framework 1.0, which will soon be supplanted by version 2.0, I propose to make the following changes:

Add a new accessor to `RigidBodyTree::sensors`:

```
const std::vector<std::shared_ptr<RigidBodySensor>> & GetSensors() const {
  return sensors;
}
```

Use Run Time Type Identification to downcast `RigidBodySensor` to the specific type of sensor, e.g., `RigidBodyDepthSensor`, `RigidBodyAccelerometer`, and `RigidBodyGyroscope`, `RigidBodyMagnetometer`. 

Finally, extract the sensor states from the output signal of `RigidBodySystem` using the sensor-specific parameters from the concrete `RigidBodySensor` objects.

liangfok, 2016-04-22:
Looking at [QuadrotorState](https://github.com/RobotLocomotion/drake/blob/master/drake/examples/Quadrotor/Quadrotor.h#L8) for an example of how higher-level semantic information can be mapped onto an Eigen Vector.

jwnimmer-tri, 2016-04-22:
I have some ideas for this, and can brief you offline.  There are two main tactics:
- Add a RBState Output struct and change RBS to emit it.
- Use an adapter view to provide API names atop the underlying Eigen output.

liangfok, 2016-04-22:
@jwnimmer-tri: thanks for the suggestions. Would you recommend manually creating the adapter based on the model or automatically generating it based on the contents of the `RigidBodySystem::sensors` vector? I'm about to implement the later and use ROS' [sensor_msgs/LaserScan](http://docs.ros.org/api/sensor_msgs/html/msg/LaserScan.html) message as the adapter for the LIDAR sensor portion of the state vector.

liangfok, 2016-04-23:
My latest plan is to not provide an accessor to `RigidBodySystem::sensors` but modify the output of `RigidBodySystem` to be a new class:

```C++
template <typename ScalarType>
class RigidBodySystemOutput {
  /**
   * The actual output of the rigid body system.
   */
  Eigen::Matrix<ScalarType, Eigen::Dynamic, 1>  output_vector_;

  /**
   * A description of the rigid body tree portion of rigid body system's output vector.
   */
  RBTDescription rigid_body_tree_description_;

  /**
   * Descriptions of the sensor outputs contained within the rigid body system's
   * output vector
   */
  std::vector<SensorDescription<ScalarType>> sensor_description;
}
```

The current implementation of `RigidBodySystem` only provides `output_vector_` in the class above. By changing the output of `RigidBodySystem`, a much richer set of semantic information can be included with the output.

```C++
/**
 * Describes the information contained within the rigid body tree's output that relates to
 * to the rigid body tree. Specifically, provides information about where the joint positions and
 * velocities can be found within the output vector of the rigid body tree.
 */ 
class RigidBodyTreeDescription {
  /**
   * The number of joint position values within the output vector of the rigid body system.
   */
  int num_position_states_;

  /**
   * The number of joint velocity values within the output vector of the rigid body system.
   */
  int num_velocitiy_states_;
}
```

`SensorDescription` is an abstract class. It is the top-level parent-class to all sensor output description classes.

```C++
class SensorDescription {
}
```

RussTedrake, 2016-04-23:
You'll want to make sure this is compatible with System 2.0

sherm1, 2016-04-23:
I like the data/metadata split but I would like to consider formalizing that in System 2.0 so that we can say that an Output (for example) consists of data + metadata where each of those is a recognizable object. That would allow us to avoid a metadata free-for-all and impose some requirements there. For example, we may want to require that metadata be self-identifying in some way to permit some introspection.

liangfok, 2016-04-24:
@sherm1: Reading [this post](http://stackoverflow.com/questions/41453/how-can-i-add-reflection-to-a-c-application) about C++'s support for reflection, it appears to require some really heavyweight pre-processor macro magic. If possible, I would like to avoid such complexity, and think we can do that by making the output signal a class that contains a map where the key is a string and the value is an object. We can create a formal [ontology](https://en.wikipedia.org/wiki/Ontology_(information_science)) of keys with specific meanings. For example, keys for the output of `RigidBodySystem` would contain two keys: "RigidBodyTreeDescription" and "SensorDescriptions".

Here's a possible design:

```C++
/**
 * An abstract superclass of all Drake signal property descriptions.
 * This contains information about a particular aspect of a signal.
 * It is the "value" within the DrakeSignal::description_ map.
 */ 
class DrakeSignalPropertyDescription {
}

/**
 * Implements a Drake signal. It may be an input, state, or output signal.
*/
template <typename ScalarType>
class DrakeSignal {
  /**
   * The actual signal.
   */
  Eigen::Matrix<ScalarType, Eigen::Dynamic, 1>  signal_;

  /**
   * A map containing details of the signal. The keys are strings that are part of Drake's
   * signal description ontology. The values are subclasses of DrakeSignalProperty.
   */
  std::map<std::string, DrakeSignalPropertyDescription> description_;
}
```

@RussTedrake: I believe the design above is compatible with the System 2.0 Framework because it essentially wraps additional metadata describing the values in System 2.0's `VectorInterface`. This meta data could then be stored within System 2.0's `Context` object. Hopefully @david-german-tri, can confirm / deny whether this is true.

RussTedrake, 2016-04-24:
System 2.0 already has the notion of a NamedVector interface, which is approximately what we have as CoordinateFrame in Matlab.  I think it captures this..?

liangfok, 2016-04-24:
Ah yes, I forgot about `NamedVectorInterface`. Looking at it, the biggest difference is it maps a string description to a `ScalarValue` whereas my proposal maps the string description to an object that is a subclass of `DrakeSignalPropertyDescription`. Clearly, I'm proposing a more powerful abstraction in terms of organizing meta data, though I agree it is likely not strictly necessary.

Intuitively, the output vector contains a hierarchy of semantics, which at first glance would be better captured by a hierarchy of objects rather than a collection of string keys each matched to a single number.

liangfok, 2016-04-24:
Thinking about this some more, the currently proposed System 2.0 `NameVectorInterface` map cannot express the name of the sensor, since the value must be a `ScalarValue`. I suppose it could contain the map "Sensor1Name" --> 2, and have another table somewhere that maps the value 2 to the actual name "RearLeftLIDAR", but this seems to indicate the currently chosen abstraction is insufficiently expressive.

RussTedrake, 2016-04-24:
In the matlab code (which is far from perfect in this regard) we have the notion of a multicoordinateframe which is a list of subframes (including other multicoordinateframes).  This has been sufficient to handle the complexity of eg multiple sensor output frames, etc.

liangfok, 2016-04-24:
Interesting. I can see how having a multi-coordinate frame would be useful in identifying where a sensor is in the world. This, I believe, it analogous to ROS' [tf](http://wiki.ros.org/tf) infrastructure.

How does the Matlab version of Drake express meta data about unique properties of the sensor? For example, how does it express the following properties of a LIDAR scanner, which I copied from [here](https://github.com/RobotLocomotion/drake/blob/master/drake/systems/plants/RigidBodySystem.h#L484)?

```
  double min_pitch;       // minimum pitch of the camera FOV in radians
  double max_pitch;       // maximum pitch of the camera FOV in radians
  double min_yaw;         // minimum yaw of the sensor FOV in radians
  double max_yaw;         // maximum yaw of the sensor FOV in radians
  size_t num_pixel_rows;  // number of points in the image vertically (pitch)
  size_t num_pixel_cols;  // number of points in the image horizontally (yaw)
  double min_range;       // minimum range of the sensor in meters
  double max_range;       // maximum range of the sensor in meters
```

sherm1, 2016-04-24:
> I would like to avoid such complexity, and think we can do that by making the output signal a class that contains a map where the key is a string and the value is an object. 

I agree. That's what I had in mind: something like a (key,value) pair where the key is a string and the value is an `AbstractValue` and can thus be any type.

RussTedrake, 2016-04-24:
It does not.  That data is not attached to the "signal" -- if you need it then you have to ask the manipulator for it's list of sensors

liangfok, 2016-04-24:
OK. I would like to advocate that System Framework 2.0 achieve a higher level of decoupling between the consumer of a signal and everything else in Drake. Ideally, a consumer of a signal should be able to do everything it needs to do using just the data contained within the signal. It should not need to also have a reference to the rigid body system, rigid body tree, etc., to get additional information to understand the meaning of the data contained within the signal. This additional decoupling will become increasingly important once we begin supporting distributed execution of Drake across many cores / PCs.

liangfok, 2016-04-24:
If the consumer of the signal, after understanding the information contained within the signal, needs to modify another component in Drake like the controller, rigid body tree, rigid body system, etc., it should be done through another signal, not directly. Again, this elevated decoupling will become important once Drake is running in a fully distributed manner, which I believe will be necessary to achieve super-high-scalability like simulating thousands of cars, pedestrians, trees, etc.

liangfok, 2016-04-24:
@sherm1: I was just thinking that a `string --> AbstractValue` mapping may not be sufficient since the consumer will be left "shooting in the dark" trying to determine the subclass of `AbtractValue` that was actually received. Maybe we should subclass `AbstractValue` using `EasilyIdentifiableAbstractValue` with a `get_type()` method that returns a type enumeration that we manually specify. This type enumeration can include an "other" type that is the catch-all fallback. If the type is "other", the consumer of the signal will need to then play the guess-the-subtype game.

sherm1, 2016-04-24:
> I was just thinking that a string --> AbstractValue mapping may not be sufficient since the consumer will be left "shooting in the dark" trying to determine the subclass of AbtractValue that was actually received. 

I don't think it is different than the current situation in that respect -- either way you need to know the expected type:
```c++
int num_values = output.get_num_values();
// vs something like
int num_values = output.GetMetaData<int>("num_values");
```
The main difference is when an error would be reported, compile time or run time. In exchange for the delayed error detection, we would get considerable flexibility (e.g. add/remove metadata w/o caller recompile, provide or withhold metadata based on runtime inputs, deal with missing metadata) and the ability to do introspection (query available metadata). I think this could provide better support for Systems as binary plug-ins also.

liangfok, 2016-04-25:
I agree either way the expected type needs to be known, but the expected type may not be at the lowest level of a class hierarchy. There are situations where the expected type needs to be subclassed by an effectively infinite number of subclasses. I'm worried about the code that will need to be written to downcast the expected type to the lowest-level of the class hierarchy to get all of the information about the signal.

Actually, thinking about this some more, I think I found a workaround for the problem I was worried about. The solution is to provide a `get_type()` method within the top-level `AbstractValue` class' templates type, not in `AbstractValue` itself.

For example, consider the output signal of `RigidBodySystem`. As mentioned above, one of its keys will be "SensorDescription". This key will map to an `AbstractValue<SensorDescription>` class. Class `SensorDescription` will have numerous sub-classes each describing a different sensor. The receiver of the signal will want to downcast `SensorDescription` to the lowest-level of the class hierarchy to get the full sensor description. To facilitate this without having to doing many guess-and-test downcasts, `SensorDescription` can provide a `get_type()` method that returns an enumeration that corresponds to a specific lowest-level child class of `SenseorDescription`.

sherm1, 2016-04-25:
AbstractValue is not a templatized type. Do you mean Value<T>?

Generally speaking any need for `get_type()`-like methods or enumerations of types indicates a bug in class architecture. (See style guide [here](https://google.github.io/styleguide/cppguide.html#Run-Time_Type_Information__RTTI_) for example: "Decision trees based on type are a strong indication that your code is on the wrong track.") Why do you think that is necessary here?

liangfok, 2016-04-25:
Sorry, I was clearly using the wrong terminology. Yes, I am referring to `Value<T>` within `AbstractValue`.

For the output of `RigidBodySystem`, the key "SensorDescription" will map to a `T` where `T` is a `SensorDescription`. However, there will actually be an infinite number of subclasses of `SensorDescription`, one for a each make/model of sensor. How does a consumer of this signal quickly downcast `SensorDescription` to a specific type of `SensorDescription`?

david-german-tri, 2016-04-25:
@liangfok I agree that System2 as written is compatible all of the proposals in this thread, since all of them can easily satisfy `VectorInterface`.  

However, I am not convinced self-description has so much universal value that we should bake it into System2.  For most simulations, including distributed ones, it seems entirely practical to load a model description at System init time.  We already have serialization formats aplenty (URDF, SDF).

liangfok, 2016-04-25:
I agree every process in the system can load the model, but it's the _structure of the system-graph_ that's currently not known by every process and what determines the semantics of the signal received by a particular system. How does a `LidarPublisher` system know which LIDAR sensors its receiving as a signal and how does it extract out the relevant information from the signal when it does not know from whom the signal originated?

david-german-tri, 2016-04-25:
@liangfok Yes, a System has to down-cast from `VectorInterface` to the specific input type it's expecting.  That type (by definition) has enough semantic richness for the System to do its work.  For instance, a `RigidBodyLIDARPublisher` could down-cast to something like your `RigidBodySystemOutput`, then iterate and publish all the LIDAR sensors.  This works just fine across processes, given typed serialization/deserialization, which is accommodated and encouraged in System2.  

Could you explain again why you want a more general typeless metadata system?

liangfok, 2016-04-25:
@david-german-tri: How does `RigidBodyLIDARPublisher` know it is being "wired" to receive the output of a `RigidBodySystem` vs. another system like `LIDARSensorSystem`?

david-german-tri, 2016-04-25:
Because that's intrinsic in its type.  If it isn't, the down-cast fails.

liangfok, 2016-04-25:
Then we will need to violate the following [Google style guide rule](https://google.github.io/styleguide/cppguide.html#Run-Time_Type_Information__RTTI_).

> Needing to know the type of an object at runtime is often an indication that the design of your class hierarchy is flawed.

david-german-tri, 2016-04-25:
Nope: we will not be querying/dispatching based on type, we will simply be throwing a hard failure if the  type is wrong.  In other words, "the logic of a program guarantees that a given instance of a base class is in fact an instance of a particular derived class", so "`dynamic_cast` may be used freely".

liangfok, 2016-04-25:
We can't simply throw a hard failure if the first try at guessing the type fails because a future guess at another type may work.

Regardless, I vote to table this discussion for now and to revisit it after I've completed the initial `LidarVisualizer` implementation, which will have direct access to the `RigidBodySystem`.

david-german-tri, 2016-04-25:
> We can't simply throw a hard failure if the first try at guessing the type fails because a future guess at
> another type may work.

It's not a guess.  A System must be wired to the correct input types.  If it is not, the entire diagram is invalid.  In System1, this is enforced at compile time via templates.  In System2, we are deferring enforcement to runtime, but we are not relaxing the requirement.

If you have multiple Systems that do similar things with different input types, you can use libraries, inheritance, composition, or templates to share code between them, but they cannot actually be the same class.

Does that make sense?  I'd rather not close this thread just yet, because I think poking at the design in this level of detail is really useful.  I'm just not yet convinced there's a hole here.

liangfok, 2016-04-25:
Suppose I have a system of type `FooSystem` that takes as input signals of class type `BarSignal`. 

Suppose I define 10 new child classes of `BarSignal` that specializes `BarSignal` in some manner.

Is there anything in System Architecture 2.0 other than the style guide that prevents a developer from modifying `FooSystem` to randomly guess-and-test down-casting `BarSignal` to each of its 10 subclasses to get the full information content contained within the subclass?

Not saying the above must, could, or should happen. Just exploring the capabilities of the proposed system.

david-german-tri, 2016-04-25:
The System2 cast utility (not yet written) will use `static_cast` in release mode, which will break type dispatch of that form.  If `FooSystem` uses RTTI directly, I don't know what we could do to prevent it, though I'd be open to suggestions.  I suppose we can have a linter rule to prevent such Systems from landing in Drake, at least.

liangfok, 2016-04-25:
Just to clarify, are you saying that once the System2's cast utility is written, `FooSystem` would not be able to receive any subclass of `BarSignal` as an input signal?

david-german-tri, 2016-04-25:
No, I'm not saying that.  The System2 cast utility will use `static_cast` in Release mode for performance reasons, which will have the beneficial side effect of making it impossible to do type dispatch (it always appears to succeed, and then segfaults).  However, we can't really prevent `FooSystem` from invoking `dynamic_cast` directly.  The best we can do is prevent such code from landing in Drake (by style guide, review, and possibly static analysis).

liangfok, 2016-04-25:
@david-german-tri: Below is the current definition of a system's output in Framework 2.0:

```
struct SystemOutput {
  // Continuous real-valued outputs from the System.
  std::vector<VectorInterface<ScalarType>> continuous_output; 

  // Discrete real-valued outputs from the System.
  std::vector<VectorInterface<ScalarType>> discrete_output;

  // Discrete non-real-valued outputs from the System.
  std::vector<AbstractValue> mode_output;
}
```
I now understand that `VectorInterface<ScalarType>` can be sub-classed and arbitrary information can be added to it. Thus, I believe it should be sufficiently expressive to contain a full description of the information contained within the signal, at all levels of abstraction (e.g., individual sensor measurement, sensor parameters, joint position, joint velocity, etc.). One thing that remains concerning to me is the fact that no structure is given for organizing all of this meta-data. I believe this is what @sherm1 meant above when he said:

> ...avoid a metadata free-for-all and impose some requirements there. For example, we may want to require that metadata be self-identifying in some way to permit some introspection.

Do you think imposing some structure on the meta data is out of scope for System Framework 2.0?

david-german-tri, 2016-04-25:
Great.  I'm not sure it's even possible to provide a useful framework for "metadata" in that sense.  The problem is so general I fear we'd just wind up inventing yet another description language.

However, yes, I think it's also out of scope for System2.  If there's a need for structured metadata, it can be implemented in a child of `VectorInterface` down the road.

liangfok, 2016-04-25:
One could argue that System 2.0 is just another component based architecture that is in fact less flexible than something like ROS since the component graph needs to be explicitly defined by the developer. In ROS, the call graph's signals are implicitly defined and automatically instantiated via the concept of "topic names." 

How would we respond to such criticism?

david-german-tri, 2016-04-25:
Isn't that really a question about Drake, not System2?  I think the party line answer is that Drake has glass-box state, with all the good things that follow from that.

liangfok, 2016-04-25:
System 2 is intrinsic to Drake so I would consider the criticism to be applicable to both. Basically, the glass-box state of Drake is orthogonal to the communication mechanism employed. We should be able to achieve glass-box state regardless of whether the signals are implemented via the transfer of sets of `VectorInterface<ScalarType>` classes, LCM messages, or ROS messages. I presently do not see the need to design / implement another component-based architecture when something like ROS already exists.

sherm1, 2016-04-25:
> I presently do not see the need to design / implement another component-based architecture when something like ROS already exists.

The model here is Simulink, not ROS. They exist for very different purposes.

liangfok, 2016-04-25:
I clearly don't understand the differences but perhaps it would for me to learn F2F with someone who knows.

liangfok, 2016-04-26:
@sherm1: I discussed this with @david-german-tri today and we concluded that the only argument against wholeheartedly embracing a component-based architecture vs. traditional object oriented architecture is latency, specifically the communication latency between components. I fully agree that the latency will be higher, but I have doubts about whether it is prohibitively higher. We agreed to at least strive to make System 2.0's architecture easily "componentized" at any time in the future. That would allow me or anyone else who's interested to try to implement a component-architecture-based Drake system to get empirical measurements on how bad the performance hit really is and hopefully prove that it's tenable for certain simulations where computation instead of communication is the long pole in the latency graph.

sherm1, 2016-04-27:
Liang, I think this "component based" conversation is orthogonal to System 2.0. System 2.0 is about building a *single* composite dynamic system (that is a term of art from the hybrid simulation community as modeled in Simulink) by composing a set of "blocks" together. This involves tight coupling between those blocks, allocation of resources from a common pool, computation of the correct execution order for the system as a whole, and a shared solver applied to the whole system to enable simultaneous solution of the coupled equations that result. At best I believe those add up to a single "component" in the sense you mean.

A full ROS-like system would consist of many independent subsystems (some of which are likely dynamic systems that could be built with System 2.0) that are loosely connected by a message-passing interface.

liangfok, 2016-04-27:
OK hopefully things will become more clear as the System 2.0 document is more fleshed out. I asked @david-german-tri to add more details about how the architecture will be _used_ to implement the key abstractions within Drake.

I'm basing my comments on the current implementation of Drake, which appears to mash the simulator, models, controllers, planners, sensors, and everything else into a single executable / library. I believe the integration of so many components into a single executable will become untenable once we start to add even more sophisticated AI components to the mix.

sherm1, 2016-04-27:
To add to the confusion, I just ran into this by accident: [Component based modeling in Simulink](http://www.mathworks.com/help/simulink/component-based-modeling.html). They mention Simulink models with 10,000 blocks.

liangfok, 2016-04-27:
Nice. 10,000 blocks is nothing 😉. Let's shoot for simulating millions of blocks each representing an intelligence sprite: https://github.com/RobotLocomotion/drake/issues/2192#issuecomment-215071260

david-german-tri, 2016-11-09:
@liangfok Since this is about System1, I would like to close it as obsolete.  If some of the same concerns apply to RigidBodyPlant, perhaps open a new issue?

closed 2016-11-09 by david-german-tri

liangfok, 2016-11-09:
Done: #4094