RobotLocomotion/drake #2154: Mac continuous build failing on gurobipy.so not found

david-german-tri, 2016-04-22:
```
[error: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/install_name_tool: can't open file: /usr/local/Cellar/python/2.7.11/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/gurobipy/gurobipy.so (No such file or directory)]
```

Since https://drake-jenkins.csail.mit.edu/view/Continuous/job/mac-clang-continuous-debug/71/

jamiesnape, 2016-04-22:
Is it unique to the CI builds?

david-german-tri, 2016-04-22:
It started after #2133.  That PR passed on Jenkins before landing, and it looks pretty unlikely to be a culprit to me.  @sammy-tri, what do you think?

jwnimmer-tri, 2016-04-22:
@david-german-tri Yes.  The python change there is never executed as part of the build system, and the header changes to relocate the pragma-once within an example shouldn't affect gurobi at all.

liangfok, 2016-04-23:
I just tried to compile Drake on OS X with Matlab and Gurobi enabled. After the build appeared to stall, I hit the enter key and got the following prompt:

[image]

Could this be related to this issue?

RussTedrake, 2016-04-23:
see https://github.com/RobotLocomotion/drake/issues/1662 <https://github.com/RobotLocomotion/drake/issues/1662>

liangfok, 2016-04-23:
I see, so only people who know the password should enable Gurobi?

RussTedrake, 2016-04-23:
Academics can make a free account.

liangfok, 2016-04-23:
OK in that case I'm guessing the need to have a Gurobi.com account is not related to this issue. Sorry for polluting this issue thread.

jamiesnape, 2016-04-25:
The CI does not use this method anyway. It has an installer package already downloaded.

david-german-tri, 2016-04-25:
Both of the PRs that landed around the time this build went red are exonerated: #2137 -> #2158, #2132 -> #2159.  Interestingly, the last successful continuous build has the same error messages, but reports as passing in Jenkins:

https://drake-jenkins.csail.mit.edu/view/Continuous/job/mac-clang-continuous-debug/70/consoleFull

david-german-tri, 2016-04-25:
Specifically, the following appears in build 71 but not build 70, even though the same error was present in both.  I don't yet know where the Regex check is coming from.

```
17:23:25 Checking log file 'build.log' for errors.
17:23:25 
17:23:25 Regex [No such file or directory] found:
17:23:25    [error: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/install_name_tool: can't open file: /usr/local/Cellar/python/2.7.11/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/gurobipy/gurobipy.so (No such file or directory)]
17:23:25    [error: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/install_name_tool: can't open file: /usr/local/Cellar/python/2.7.11/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/gurobipy/gurobipy.so (No such file or directory)]
17:23:25 setting the build status to FAILURE
```

jamiesnape, 2016-04-25:
Nothing changed at all as far as CI is concerned. The regex check is just reporting errors found earlier in the build. It is not causing the error.

david-german-tri, 2016-04-25:
OK, so again, why did the regex check report errors in #71 but not #70, when the same errors are present in both?

https://drake-jenkins.csail.mit.edu/view/Continuous/job/mac-clang-continuous-debug/70/consoleFull
https://drake-jenkins.csail.mit.edu/view/Continuous/job/mac-clang-continuous-debug/71/consoleFull

david-german-tri, 2016-04-25:
I will bet dollars to doughnuts this is what changed as far as CI is concerned:

https://github.com/billhoffman/drake/commit/59859240607fc5a64525aa7e18dbd592528eb26b

jwnimmer-tri, 2016-04-26:
Bumped priority due to PR merging headaches with to the always-red OSX builder.

jamiesnape, 2016-04-26:
Working on a script that no longer needs to parse the output log. Should be deployed by lunchtime all being well.

david-german-tri, 2016-04-26:
I'm guessing that was https://github.com/billhoffman/drake/commit/5bf065d61e8c8cb6c341f2c54926a884080bcfbe#diff-d6c5855a62cf32a4dadbc2831f0f295f and https://github.com/billhoffman/drake/commit/ad0254f35d4b628c3f951da4a261a8e152fac31b#diff-d6c5855a62cf32a4dadbc2831f0f295f.  

Can "drake_jenkins_ctest2.cmake" be deleted?  Seems to be identical to ctest_driver_script.cmake.

closed 2016-04-26 by david-german-tri

jamiesnape, 2016-04-26:
Eventually once the builds have cycled through.