RobotLocomotion/drake #2169: HybridDrakeSystem get state frame for each mode

hongkai-dai, 2016-04-24:
Currently when we call `getStateFrame` for a `HybridDrakeSystem` (for example, a SLIP model), what it returns is the state frame of mode 1. This has a problem if different modes have a different state frames. Alternatively, we can get the state frame of each of the modes, using `getMode(mode_num).getStateFrame`, but what it returns does not include the discrete state (the mode number). I believe the correct solution is to overload the `getStateFrame/getOutputFrame` method for `HybridDrakeSystem`, and pass the mode number as the input argument to those functions. So
```
SLIP.getStateFrame(stance_mode_number)
```
will return a frame like `[stance_mode_number;r;theta;r_dot;theta_dot;x_toe]`, with 6 states
```
SLIP.getStateFrame(flight_mode_number)
```
will return a frame like `[flight_mode_number;x;y;xdot;ydot]`, with 5 states

RussTedrake, 2016-04-25:
Seems reasonable to me.

hongkai-dai, 2016-09-08:
@RussTedrake , this was a problem in the MATLAB code, do we want to fix the MATLAB code?

RussTedrake, 2016-09-08:
if it’s not blocking anybody and is not a trivial fix, then let’s resolve this as “won’t fix”.

hongkai-dai, 2016-10-26:
won't fix

closed 2016-10-26 by hongkai-dai