RobotLocomotion/drake #2170: Errors in CompassGait examples

boxgf, 2016-04-25:
Hi all, 

I have a few questions regarding CompassGaitPlant.

1) The overloaded manipulatorDynamics() function in CompassGaitStancePlant.m returns B = [-1;1]. Since we do not actuate the stance leg with respect to ground, shouldn't it be B = [-1;0]? 

2) The footCollisionGuard2 I guess we should be checking theta_st + gamma > 0. Using only theta_st  > 0 may prevent a collision check for high values of gamma

3) The sampledFiniteTimeVerification seems to always throw Drake:PolynomialTrajectorySystem:InfeasibleRho error for both a GaitPlant (using only one mode i.e, p.modes{1}) and an Acrobot.

Once I find the cost to go evaluated for a trajectory, I input the cost to go at the last time step as G (the parent spot lyapunov function) just to check if the sampledFiniteTimeVerification function works fine. I still get the same error. Am I missing something here. Is there a sample code where the funnel verification is done? 

Thanks in advance.

RussTedrake, 2016-04-25:
A few quick answers (without looking back at the code)
> 1) The overloaded manipulatorDynamics() function in CompassGaitStancePlant.m returns B = [-1;1]. Since we do not actuate the stance leg with respect to ground, shouldn't it be B = [-1;0]?
> 
No.  The angles are in world coordinates, not relative, so the B is correct.

> 2) The footCollisionGuard2 I guess we should be checking theta_st + gamma > 0. Using only theta_st > 0 may prevent a collision check for high values of gamma
> 

I believe this is just the foot scuffing avoidance condition, making sure that we don't trigger unless the stance leg has passed vertical.
> 3) The sampledFiniteTimeVerification seems to always throw Drake:PolynomialTrajectorySystem:InfeasibleRho error for both a GaitPlant (using only one mode i.e, p.modes{1}) and an Acrobot.
>
> Once I [...]

You do have to (potentially manually) find an initial rho that is feasible.  Yes -- there are a number if samples in the example directories.  Just search for calls to that function and you'll find them.

boxgf, 2016-04-25:
Thank you very much professor.

closed 2016-05-19 by david-german-tri