RobotLocomotion/drake #2175: Unit test drake/systems/plants/test/compareParsers.m Should Use Explicit List of URDF Files

liangfok, 2016-04-25:
Currently, it grabs every single URDF in `drake/systems/plants/test/`, even those that it should not because they, for example, contain a world link. This causes the unit test to fail when it should not.

Example: https://drake-cdash.csail.mit.edu/testDetails.php?test=510683&build=11999

This is causing CI failure of https://github.com/RobotLocomotion/drake/pull/2171.

liangfok, 2016-04-25:
@RussTedrake, since this is a Matlab code change, I'll assign this to you.

liangfok, 2016-04-26:
Looking at the top of `drake/systems/plants/test/compareParsers.m`, I see that it can take a list of URDF files as an input parameter and only grabs all of the URDF files if no such list is provided.

```
function compareParsers(urdfs)

if nargin<1 || isempty(urdfs)
  urdfs = allURDFs();
elseif ~iscell(urdfs)
  urdfs = {urdfs};
end
```
Now to figure out how to supply it a list of URDFs...

liangfok, 2016-04-26:
Looking in `drake/systems/plants/test/CMakeLists.txt`, I see the following line that adds the compareParsers.m unit test to the list of tests to run:

```
add_matlab_test(NAME systems/plants/test/compareParsers COMMAND compareParsers)
```

Thus, I'm now wondering whether `add_matlab_test()` can accept as a parameter the list of URDF files to evaluate.

jwnimmer-tri, 2016-04-26:
Random idea for your consideration...

If the general premise is that ~most URDF files are supposed to succeed, but sometimes implementation deficiencies prevent certain ones from succeeding, than a test blacklist rather than a test whitelist may be a more durable solution.

The blacklist would say "disable testing of foo, because the world frame is unimplemented see #abc).

By having a whitelist, you have to be careful that people who add models also add it to the list of test.

Of course, this assumes that models in that directory are generally supposed to be all tested.

liangfok, 2016-04-26:
@jwnimmer-tri: Nice! Thanks for the advice. I think I will go with it since, yes, most URDFs are supposed to pass. It's only failing from the two new URDFs that I added to test https://github.com/RobotLocomotion/drake/pull/2171.

I believe it's because I updated the C++ URDF parser but not the Matlab parser (it would take me an unknown amount of time to learn how to update the Matlab URDF parser)

liangfok, 2016-04-27:
I tried to replicate this problem on OS X and ran into a problem where the call to `allURDFs()` failed with the following problem:

```
$ ctest -VV -R systems/plants/test/compareParsers
UpdateCTestConfiguration  from :/Users/liang/dev/drake/drake/pod-build/DartConfiguration.tcl
Parse Config file:/Users/liang/dev/drake/drake/pod-build/DartConfiguration.tcl
UpdateCTestConfiguration  from :/Users/liang/dev/drake/drake/pod-build/DartConfiguration.tcl
Parse Config file:/Users/liang/dev/drake/drake/pod-build/DartConfiguration.tcl
Test project /Users/liang/dev/drake/drake/pod-build
Constructing a list of tests
Done constructing a list of tests
Checking test dependency graph...
Checking test dependency graph end
test 147
    Start 147: systems/plants/test/compareParsers

147: Test command: /usr/local/Cellar/cmake/3.5.0/bin/cmake "-Dtest_name=systems/plants/test/compareParsers" "-Dadditional_paths=/Users/liang/dev/drake/build/matlab;/Users/liang/dev/drake/drake/systems/plants/test" "-Dtest_timeout=-1" "-Doutput_directory=/Users/liang/dev/drake/drake/pod-build/Matlab" "-Dworking_directory=/Users/liang/dev/drake/drake/systems/plants/test" "-DMatlab_PROGRAM=/Applications/MATLAB_R2016a.app/bin/matlab" "-Dno_unittest_framework=FALSE" "-DMatlab_ADDITIONNAL_STARTUP_OPTIONS=" "-Dunittest_file_to_run=" "-Dcustom_Matlab_test_command=try, eval('compareParsers'); catch ex, disp(getReport(ex,'extended')); disp(' '); force_close_system; exit(~strncmp(ex.identifier,'Drake:MissingDependency',23)); end; force_close_system; exit(0)" "-Dcmd_to_run_before_test=addpath_drake; global g_disable_visualizers; g_disable_visualizers=true" "-P" "/Users/liang/dev/drake/drake/cmake/MatlabTestsRedirect.cmake"
147: Test timeout computed to be: 1500
147: Matlab test  output:
147:
147:                             < M A T L A B (R) >
147:                   Copyright 1984-2016 The MathWorks, Inc.
147:                    R2016a (9.0.0.341360) 64-bit (maci64)
147:                              February 11, 2016
147:
147:
147: To get started, type one of these: helpwin, helpdesk, or demo.
147: For product information, visit www.mathworks.com.
147:
147:
147:    Home License -- for personal use only.  Not for government,
147:    academic, research, commercial, or other organizational use.
147:
147: Error using allURDFs (line 20)
147: still need to implement a version that works without find
147:
147: Error in compareParsers (line 8)
147:   urdfs = allURDFs();
147:
147:
147:
1/1 Test #147: systems/plants/test/compareParsers ...   Passed    8.61 sec

The following tests passed:
        systems/plants/test/compareParsers

100% tests passed, 0 tests failed out of 1

Total Test time (real) =   8.65 sec
```

Not that even though the call to `allURDFs()` failed, the unit test itself succeeded.

liangfok, 2016-05-15:
I'm going to close this issue because we've decided to use a blacklist strategy.

closed 2016-05-15 by liangfok