RobotLocomotion/drake #2176: Superbuild nlopt dependencies seem wrong

jwnimmer-tri, 2016-04-25:
I admit to having no idea how anything in the superbuild is supposed to work, but perhaps [CMakeLists.txt:L136](https://github.com/RobotLocomotion/drake/blob/136e5f487bf7f0cff7ffb69c3595f59ed70d4e1b/CMakeLists.txt#L136) is wrong?  It does not list nlopt as dependency.

The symptom I am seeing is: my drake-distro is in a state where it downloads nlopt and has it in the list-of-things-to-superbuild, but doesn't think that drake should depend on it, so compiles drake/drake first, and only ​then​ installs nlopt:
```
-- Build files have been written to: /home/jwnimmer/jwnimmer-tri/drake/drake/pod-build
[ 93%] Performing build step for 'drake'
...
[ 95%] Completed 'drake'
[ 96%] Built target drake
[ 96%] No configure step for 'nlopt'
[ 97%] Performing build step for 'nlopt'
```

The build command line was:
```
cd ~/jwnimmer-tri/drake && env PATH=/usr/lib/ccache:$PATH CXX=g++-4.9 CC=gcc-4.9 make
```

jwnimmer-tri, 2016-04-25:
In the continuous CI build, for example, it looks to me like both `nlopt` and `drake` are being compiled in parallel, and it only happens to work because `nlopt` finishes its `make install` before `drake` reaches the part of its build that requires `-lnlopt` to exist.

jwnimmer-tri, 2016-04-25:
Also FWIW, I would be a lot more confident in my understanding of how to fix the nlopt dependency if the `set(drake_dependencies, ...` line listed every dependency except for nlopt.  As it stands now, there appear to me to be many items missing, which means I must be misunderstanding something about how its supposed to work.

liangfok, 2016-04-29:
I just ran into this problem on an Ubuntu system. Worked around it by using `make` instead of `make -j8`, which is obviously not a good solution.

sammy-tri, 2016-04-29:
If we think an acceptable fix is to add nlopt as an explicit drake dependency (which does seem to cause nlopt to reliably build before drake for me), I'll take this issue as a fix to go in with some other NLopt fixes I'm preparing a PR for.  Please reassign to me if this seems reasonable.

liangfok, 2016-04-29:
Sounds reasonable to me.

sammy-tri, 2016-05-03:
Closed by #2243

closed 2016-05-03 by sammy-tri