RobotLocomotion/drake #2177: Export preprocessor symbols should be excluded from Doxygen

sherm1, 2016-04-25:
Drake's build system generates Microsoft export macros like `DRAKEUTIL_EXPORT` to force symbols to be visible in `.dll` shared libraries. Doxygen dutifully includes these ugly and meaningless things in the generated documentation. They should not be included.

Normally special macro handling is done by modifying `Doxyfile` which has provision for doxygen-specific macro expansion or elimination. That would be hard to do nicely with build-time generated export symbols. However, the symbols could exclude themselves from Doxygen in the generated export files using a symbol that is only present when Doxygen is running (doxygen manual suggests `DOXYGEN_SHOULD_SKIP_THIS`; see the `PREDEFINED` keyword for Doxyfile).

FYI, code can be skipped at point of use with
```
/** @cond **/ 
Doxygen won't see this.
/** @endcond **/
```
(The space after `@cond` is required for some reason.) So this works but is awkward:
```/** @cond **/ DRAKEUTIL_EXPORT /** @endcond **/```

sherm1, 2016-04-27:
@jwnimmer-tri discovered [this relevant doxygen manual page](https://www.stack.nl/~dimitri/doxygen/manual/preprocessing.html). There might be a good hint there.

jwnimmer-tri, 2016-04-27:
Yeah.  My thought was just to list all of the FOO_EXPORT symbols that we know about in the Doxyfile.  It's a bit brittle, but if we miss one it's quickly fixed once we notice, which seems like a fair compromise for generated documentation.

PR #2994 (2016-08-01): Add doxygen suppressions for _EXPORT goop
Closes #2177.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/2994)
closed 2016-08-01 by jwnimmer-tri