RobotLocomotion/drake #2178: Consider using WINDOWS_EXPORT_ALL_SYMBOLS

jwnimmer-tri, 2016-04-25:
See [Create dlls on Windows without declspec() using new CMake export all feature](https://blog.kitware.com/create-dlls-on-windows-without-declspec-using-new-cmake-export-all-feature/).  Possibly this would be a good idea for Drake to adopt?

Right now we are restricting linking on msvc builds only (and not on Linux or OSX), and I'm not sure if that really has any advantage, but can causing frustration for developers who don't get the magic spells correct prior to CI.

This would make all the FOO_EXPORT (for ~20 values of FOO) junk in the tree disappear, and be a different way of fixing #2177.

This depends on #2082 of course.

sherm1, 2016-04-27:
> This would make all the FOO_EXPORT (for ~20 values of FOO) junk in the tree disappear, and be a different way of fixing #2177.

That would be lovely! But I am skeptical that it will actually have that effect. As I read the new CMake feature, it fails to deal with static data in exported classes. So any classes that have static data members will still need to have export headers generated and then the code has to be modified to insert them. I am not sure that's an improvement over the current situation where there is a conventional rote method of shoving these ugly export symbols in front of every class.

jwnimmer-tri, 2016-04-27:
I am also skeptical, but willing to give it a try and see?  Though if templated classes don't work, that may be a showstopper for Drake.

Really, I'd be happy enough if we had one DRAKE_EXPORT coming from `common` somewhere, and we used that on every class by rote, instead of having ~20 varieties of CMake-generated export headers that are all identical anyway.

sherm1, 2016-04-27:
> I'd be happy enough if we had one DRAKE_EXPORT coming from common somewhere, and we used that on every class by rote ...

(I'm not defending this shitstorm!) The reason they are different is that the macros expand differently when we are *building* a library than when we are *using* that library. While building `libdrakeY` we make use of headers whose definitions are in `libdrakeX`. At that moment, DRAKEX_EXPORT expands to `dllimport` while DRAKEY_EXPORT expands to `dllexport`. Hence they must be different macros :cry:.

jwnimmer-tri, 2016-04-27:
I have an existence proof that the same `#include` statement and macro name can work everywhere.  I will ping you in direct email about it.

billhoffman, 2016-04-27:
I would like to see the existence proof of the same include working, maybe you could post a link to an example github repo that shows this being done.  As was mentioned you need it to be different when building than when using the library.

jwnimmer-tri, 2016-04-27:
So @sherm1 and I talked this over in email with my code example.  (Sorry I can't share.)

The gist was (1) when building my ~20 DLLs, the headers always preprocess to say dllexport; (2) when building programs that use those DLLs, the headers always preprocess to say dllimport.  This is correct and typical for the program-calls-library scenarios (where one translation unit saw export, and one saw import), but unusual for the library-calls-library scenarios (where they both saw export).

Apparently this can work for many or most cases, but per [this blog post](https://blogs.msdn.microsoft.com/russellk/2005/03/20/lnk4217/), my code would have been less performant, because it would be using more thunking when the libraries call into each other, versus the carefully-crafted import/export naming as is more typically done.  The thunks may not have been being removed.

Bottom like, that trick is not appropriate to Drake, where calls between the libraries within drake need to be performant.

jwnimmer-tri, 2016-04-27:
So in short, sorry for the detour!

I still think it'd be informative to take ALL_SYMBOLS for a spin, though not a current priority.

jwnimmer-tri, 2016-07-30:
Unlikely to ever amount to anything.  Closing.

closed 2016-07-30 by jwnimmer-tri