RobotLocomotion/drake #2179: Unable to run drake-visualizer on Ubuntu Linux

liangfok, 2016-04-25:
Here is the error:

```
$ ./build/bin/drake-visualizer 
['/home/liang/dev/drake/build/lib/python2.7/dist-packages', '/home/liang/dev/drake/build/lib/python2.7/site-packages']
Traceback (most recent call last):
 File "<string>", line 1, in <module>
 File "/home/liang/dev/drake/build/lib/python2.7/dist-packages/director/drakevisualizer.py", line 637, in main
   app = DrakeVisualizerApp()
 File "/home/liang/dev/drake/build/lib/python2.7/dist-packages/director/drakevisualizer.py", line 500, in __init__
   self.drakeVisualizer = DrakeVisualizer(self.view)
 File "/home/liang/dev/drake/build/lib/python2.7/dist-packages/director/drakevisualizer.py", line 286, in __init__
   self.sendStatusMessage('loaded')
 File "/home/liang/dev/drake/build/lib/python2.7/dist-packages/director/drakevisualizer.py", line 365, in sendStatusMessage
   msg = lcmdrake.lcmt_viewer_command()
AttributeError: 'module' object has no attribute 'lcmt_viewer_command'
```

I've verified that `drake/lcmt_viewer_command.py` exists in my `$PYTHONPATH`:

```
$ echo $PYTHONPATH
/home/liang/dev/drake/build/lib/python2.7/dist-packages:/home/liang/dev/drake/build/lib/python2.7/site-packages:/opt/ros/indigo/lib/python2.7/dist-packages
```

liangfok, 2016-04-25:
The problem is `[drake distro]/build/lib/python2.7/dist-packages/drake/__init__.py` is empty. Since this file is generated by `lcm-gen`, it indicates a problem while generating LCM messages.

jwnimmer-tri, 2016-04-25:
FWIW, I saw that this morning as well.  I assumed that I PEBKAC'd, so I retreated and re-ran the superbuild, and it started working again.

I wonder if its the same cause as #2176 -- that `set(drake_dependencies...` is missing an entry for director?  So you have to run the superbuild twice for drake to see director?

liangfok, 2016-04-25:
Yes, I think there must be something like #2176 going wrong. I've tried nuking my system numerous times now and am unable to overcome the problem of `[drake distro]/build/lib/python2.7/dist-packages/drake/__init__.py` being empty. Looking through the build log messages, I see the lines below that seem to indicate `__init.py` is being installed and is up to date, but the actual content of the file is empty.

```
-- Installing: /home/liang/dev/drake/build/lib/python2.7/dist-packages/drake/__init__.py
-- Installing: /home/liang/dev/drake/build/lib/python2.7/dist-packages/drake/lcmt_support_data.py
-- Up-to-date: /home/liang/dev/drake/build/lib/python2.7/dist-packages/drake/__init__.py
-- Installing: /home/liang/dev/drake/build/lib/python2.7/dist-packages/drake/lcmt_robot_state.py
-- Up-to-date: /home/liang/dev/drake/build/lib/python2.7/dist-packages/drake/__init__.py
-- Installing: /home/liang/dev/drake/build/lib/python2.7/dist-packages/drake/lcmt_viewer_draw.py
-- Up-to-date: /home/liang/dev/drake/build/lib/python2.7/dist-packages/drake/__init__.py
-- Installing: /home/liang/dev/drake/build/lib/python2.7/dist-packages/drake/lcmt_qp_controller_input.py
-- Up-to-date: /home/liang/dev/drake/build/lib/python2.7/dist-packages/drake/__init__.py
-- Installing: /home/liang/dev/drake/build/lib/python2.7/dist-packages/drake/lcmt_viewer_load_robot.py
-- Up-to-date: /home/liang/dev/drake/build/lib/python2.7/dist-packages/drake/__init__.py
-- Installing: /home/liang/dev/drake/build/lib/python2.7/dist-packages/drake/lcmt_euler_floating_joint_state_t.py
-- Up-to-date: /home/liang/dev/drake/build/lib/python2.7/dist-packages/drake/__init__.py
-- Installing: /home/liang/dev/drake/build/lib/python2.7/dist-packages/drake/lcmt_piecewise_polynomial.py
-- Up-to-date: /home/liang/dev/drake/build/lib/python2.7/dist-packages/drake/__init__.py
-- Installing: /home/liang/dev/drake/build/lib/python2.7/dist-packages/drake/lcmt_quadrotor_output_t.py
-- Up-to-date: /home/liang/dev/drake/build/lib/python2.7/dist-packages/drake/__init__.py
-- Installing: /home/liang/dev/drake/build/lib/python2.7/dist-packages/drake/lcmt_simulation_command.py
-- Up-to-date: /home/liang/dev/drake/build/lib/python2.7/dist-packages/drake/__init__.py
-- Installing: /home/liang/dev/drake/build/lib/python2.7/dist-packages/drake/lcmt_zmp_com_observer_state.py
-- Up-to-date: /home/liang/dev/drake/build/lib/python2.7/dist-packages/drake/__init__.py
-- Installing: /home/liang/dev/drake/build/lib/python2.7/dist-packages/drake/lcmt_polynomial.py
-- Up-to-date: /home/liang/dev/drake/build/lib/python2.7/dist-packages/drake/__init__.py
-- Installing: /home/liang/dev/drake/build/lib/python2.7/dist-packages/drake/lcmt_foot_flag.py
-- Up-to-date: /home/liang/dev/drake/build/lib/python2.7/dist-packages/drake/__init__.py
-- Installing: /home/liang/dev/drake/build/lib/python2.7/dist-packages/drake/lcmt_driving_control_cmd_t.py
-- Up-to-date: /home/liang/dev/drake/build/lib/python2.7/dist-packages/drake/__init__.py
-- Installing: /home/liang/dev/drake/build/lib/python2.7/dist-packages/drake/lcmt_viewer_geometry_data.py
-- Up-to-date: /home/liang/dev/drake/build/lib/python2.7/dist-packages/drake/__init__.py
-- Installing: /home/liang/dev/drake/build/lib/python2.7/dist-packages/drake/lcmt_viewer_command.py
-- Up-to-date: /home/liang/dev/drake/build/lib/python2.7/dist-packages/drake/__init__.py
-- Installing: /home/liang/dev/drake/build/lib/python2.7/dist-packages/drake/lcmt_joint_pd_override.py
-- Up-to-date: /home/liang/dev/drake/build/lib/python2.7/dist-packages/drake/__init__.py
-- Installing: /home/liang/dev/drake/build/lib/python2.7/dist-packages/drake/lcmt_driving_command_t.py
-- Up-to-date: /home/liang/dev/drake/build/lib/python2.7/dist-packages/drake/__init__.py
-- Installing: /home/liang/dev/drake/build/lib/python2.7/dist-packages/drake/lcmt_zmp_data.py
-- Up-to-date: /home/liang/dev/drake/build/lib/python2.7/dist-packages/drake/__init__.py
-- Installing: /home/liang/dev/drake/build/lib/python2.7/dist-packages/drake/lcmt_quadrotor_input_t.py
-- Up-to-date: /home/liang/dev/drake/build/lib/python2.7/dist-packages/drake/__init__.py
-- Installing: /home/liang/dev/drake/build/lib/python2.7/dist-packages/drake/lcmt_whole_body_data.py
-- Up-to-date: /home/liang/dev/drake/build/lib/python2.7/dist-packages/drake/__init__.py
-- Installing: /home/liang/dev/drake/build/lib/python2.7/dist-packages/drake/lcmt_body_motion_data.py
-- Up-to-date: /home/liang/dev/drake/build/lib/python2.7/dist-packages/drake/__init__.py
-- Installing: /home/liang/dev/drake/build/lib/python2.7/dist-packages/drake/lcmt_drake_signal.py
-- Up-to-date: /home/liang/dev/drake/build/lib/python2.7/dist-packages/drake/__init__.py
-- Installing: /home/liang/dev/drake/build/lib/python2.7/dist-packages/drake/lcmt_force_torque.py
-- Up-to-date: /home/liang/dev/drake/build/lib/python2.7/dist-packages/drake/__init__.py
-- Installing: /home/liang/dev/drake/build/lib/python2.7/dist-packages/drake/lcmt_viewer_link_data.py
-- Up-to-date: /home/liang/dev/drake/build/lib/python2.7/dist-packages/drake/__init__.py
-- Installing: /home/liang/dev/drake/build/lib/python2.7/dist-packages/drake/lcmt_scope_data.py
-- Up-to-date: /home/liang/dev/drake/build/lib/python2.7/dist-packages/drake/__init__.py
-- Installing: /home/liang/dev/drake/build/lib/python2.7/dist-packages/drake/lcmt_polynomial_matrix.py
-- Up-to-date: /home/liang/dev/drake/build/lib/python2.7/dist-packages/drake/__init__.py
-- Installing: /home/liang/dev/drake/build/lib/python2.7/dist-packages/drake/lcmt_body_wrench_data.py
-- Up-to-date: /home/liang/dev/drake/build/lib/python2.7/dist-packages/drake/__init__.py
-- Installing: /home/liang/dev/drake/build/lib/python2.7/dist-packages/drake/lcmt_simple_car_state_t.py
-- Up-to-date: /home/liang/dev/drake/build/lib/python2.7/dist-packages/drake/__init__.py
-- Installing: /home/liang/dev/drake/build/lib/python2.7/dist-packages/drake/lcmt_simple_car_config_t.py
```

liangfok, 2016-04-25:
For reference, here's what `[drake distro]/build/lib/python2.7/dist-packages/drake/__init__.py` looks like on my OS X machine, which is able to run `drake-visualizer`:

```
"""LCM package __init__.py file
This file automatically generated by lcm-gen.
DO NOT MODIFY BY HAND!!!!
"""

from .lcmt_body_motion_data import lcmt_body_motion_data
from .lcmt_body_wrench_data import lcmt_body_wrench_data
from .lcmt_drake_signal import lcmt_drake_signal
from .lcmt_driving_control_cmd_t import lcmt_driving_control_cmd_t
from .lcmt_foot_flag import lcmt_foot_flag
from .lcmt_force_torque import lcmt_force_torque
from .lcmt_joint_pd_override import lcmt_joint_pd_override
from .lcmt_piecewise_polynomial import lcmt_piecewise_polynomial
from .lcmt_polynomial import lcmt_polynomial
from .lcmt_polynomial_matrix import lcmt_polynomial_matrix
from .lcmt_qp_controller_input import lcmt_qp_controller_input
from .lcmt_quadrotor_input_t import lcmt_quadrotor_input_t
from .lcmt_quadrotor_output_t import lcmt_quadrotor_output_t
from .lcmt_robot_state import lcmt_robot_state
from .lcmt_scope_data import lcmt_scope_data
from .lcmt_simulation_command import lcmt_simulation_command
from .lcmt_support_data import lcmt_support_data
from .lcmt_viewer_command import lcmt_viewer_command
from .lcmt_viewer_draw import lcmt_viewer_draw
from .lcmt_viewer_geometry_data import lcmt_viewer_geometry_data
from .lcmt_viewer_link_data import lcmt_viewer_link_data
from .lcmt_viewer_load_robot import lcmt_viewer_load_robot
from .lcmt_whole_body_data import lcmt_whole_body_data
from .lcmt_zmp_com_observer_state import lcmt_zmp_com_observer_state
from .lcmt_zmp_data import lcmt_zmp_data
from .lcmt_driving_command_t import lcmt_driving_command_t
from .lcmt_euler_floating_joint_state_t import lcmt_euler_floating_joint_state_t
from .lcmt_simple_car_config_t import lcmt_simple_car_config_t
from .lcmt_simple_car_state_t import lcmt_simple_car_state_t
```

liangfok, 2016-04-25:
Update: switched to a completely different workspace and could not replicate the problem. I've only tried compiling once so far in the new workspace so I don't know if it's coincidental luck just yet.

liangfok, 2016-04-26:
Update: Switched to the master branch and `[drake distro]/build/lib/python2.7/dist-packages/drake/__init__.py` is being correctly generated. Thus, there must be something wrong with my [feature/ros_system](https://github.com/liangfok/drake/tree/feature/ros_system) branch.

Whew! At least I'm beginning to narrow down the problem.

liangfok, 2016-04-26:
I've nailed down the bug. It's due to a conflict between the ROS and the LCM message generators. Specifically, adding the following line in `drake/CMakeLists.txt`, which invokes ROS' message generator breaks LCM's message generator.

```C++
  # Generates the code for Drake's custom ROS messages.
  generate_messages(DEPENDENCIES std_msgs)
```

:sob: 

@jwnimmer-tri  and @david-german-tri: This is another reason to adopt a more component-oriented approach. One component can be `drake_lcm_integration` while another can be `drake_ros_integration`. They will have different CMakeLists.txt that won't conflict. 

Since I don't immediately need custom messages, I'm going to kick this can down the road in issue: https://github.com/RobotLocomotion/drake/issues/2187

closed 2016-04-26 by liangfok