RobotLocomotion/drake #2181: Define GTEST_DONT_DEFINE_TEST

david-german-tri, 2016-04-25:
As discussed on #2172.

jwnimmer-tri, 2016-05-11:
Once this is done, also remove the "_Use the `GTEST` prefix in unit test declarations_" rule from `code_style_guide.rst`, since it will be impossible to do otherwise.

jwnimmer-tri, 2016-05-23:
If we are going to defect this in inspections, we better darn well just tell the compiler to pre-enforce it.  I'll implement this now.

liangfok, 2016-05-23:
Would resolving this issue involve defining a header file with this macro that is included by all Google-Test-based unit tests? In which case, maybe the same header file can also include `gtest/gtest.h`?

jwnimmer-tri, 2016-05-23:
My current spike test sets it via CMake, which is more likely to succeed than relying on using-code to remember to set it.

liangfok, 2016-05-23:
Interesting. I did not know CMake can inject macros into C++ source code. Looking forward to learning how that's done.

jwnimmer-tri, 2016-05-23:
I've pushed the fixes to github, but it hasn't deigned to show them in the web UI yet.  Once it does, I'll PR this.

rpoyner-tri, 2016-05-23:
As of recently, here is the complete list of suspect names that GTEST lets you suppress. Consider forbidding some or all of these.
<pre>
$ grep DONT ~/checkout/drake/externals/googletest/googletest/include/gtest/gtest.h
#if !GTEST_DONT_DEFINE_FAIL
#if !GTEST_DONT_DEFINE_SUCCEED
// Define macro GTEST_DONT_DEFINE_ASSERT_XY to 1 to omit the definition of
#if !GTEST_DONT_DEFINE_ASSERT_EQ
#if !GTEST_DONT_DEFINE_ASSERT_NE
#if !GTEST_DONT_DEFINE_ASSERT_LE
#if !GTEST_DONT_DEFINE_ASSERT_LT
#if !GTEST_DONT_DEFINE_ASSERT_GE
#if !GTEST_DONT_DEFINE_ASSERT_GT
#if !GTEST_DONT_DEFINE_TEST
</pre>

sherm1, 2016-05-23:
Would be good to use only the `GTEST_` form of all of these but could be a different PR. (Would be even better if Google had had some respect for the sanctity of our namespaces in the first place!)

jwnimmer-tri, 2016-05-23:
I asked rpoyner-tri to list all of the remaining ones here.  The PR will stay as-is, but I intend to keep the issue open to track the fixes to all of the pollution.

jwnimmer-tri, 2016-05-24:
I propose that we keep the ASSERT_EQ and friends (to match EXPECT_EQ), but that we nuke FAIL and SUCCEED.

PR #2431 (2016-05-24): Define GTEST_DONT_DEFINE_TEST
Closes #2181.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/2431) &emsp; Multiple assignees:&emsp;[image]&nbsp;<a href="/robotlocomotion/drake/pulls/assigned/jwnimmer-tri">jwnimmer-tri</a>,&emsp;[image]&nbsp;<a href="/robotlocomotion/drake/pulls/assigned/sherm1">sherm1</a>
commit 07ece3f4 (2016-05-24): Define GTEST_DONT_DEFINE_TEST.
Closes #2181.
PR #2443 (2016-05-25): Set GTEST_DONT_DEFINE_{FAIL,SUCCEED}=1
Closes #2181.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/2443)
commit e3e85286 (2016-05-25): Set GTEST_DONT_DEFINE_{FAIL,SUCCEED}=1.
Closes #2181.
closed 2016-05-25 by david-german-tri