RobotLocomotion/drake #2182: Poll: Where to put test files?

david-german-tri, 2016-04-25:
Two comments follow, thumbs-up the one with which you agree more.

david-german-tri, 2016-04-25:
`drake/foo/test/foo_test.cc`

david-german-tri, 2016-04-25:
`drake/foo/foo_test.cc`

jwnimmer-tri, 2016-04-26:
Explaining my vote:

In many projects, I prefer the former (a separate test directory) to help keep filecount per directory lower.  It also facilities creating test libraries that live next to the tests, but are more clearly partitioned from production code.  (So you have `libcommon` source files in e.g. `common/*.cc`, `libcommon_test` source files in `common/test/*.cc`, and test program source files also in `common/test/*_test.cc` but the `_test.cc` suffix distinguishes them.  Then, you have e.g. `libsystem_test` code and/or other tests in `system/test/` use `libcommon_test`, but `libsystem` is not allowed to use `libcommon_test`.)

However, I've always had a fixed directory tree depth (all libraries are laid out as top-level siblings, so mindepth=maxdepth=1 for production code, mindepth=maxdepth=2 for test code).  In other words, there were directories `foo` and `foo/test`, `bar` and `bar/test`, etc.  There was not `foo/test` and `foo/bar/test` and `foo/bar/baz/test` like we have with `systems/test` vs `systems/plants/joints/test` and etc., where the test subdirs are intermingled with sublibrary directories.  If we keep the deep tree depth like we have now, I think we should stop using test as a separate subdir, to better group the tests with the code they are testing.  If we flatten the tree to put all libraries at the top level, I'd vote to keep tests in subdirs (but happy to live with being outvoted).

liangfok, 2016-04-26:
Since I believe the ideal software architecture that Drake should eventually adopt is [component-based](https://en.wikipedia.org/wiki/Component-based_software_engineering), I hope Dake's deep directory structure can eventually be simplified to be:

- `component1/`
- `component1/test/`
- `component2/`
- `component2/test/`
- ...

I don't see how component-based architectures prevent Drake having a glass box state since that state can be configured to be included with the messages between components.

Maybe we should table this decision regarding location of unit tests to the more fundamental question of whether we adopt a component-based architecture.

Related: https://github.com/RobotLocomotion/drake/issues/2152#issuecomment-214556160

david-german-tri, 2016-04-26:
Explaining my vote, since my rationale is pretty different from @jwnimmer-tri: I far more often think "hey, I need to look at the tests to understand this file" than "hey, I need to look at all the tests for this directory".  Therefore, it's useful to have the tests right next to the file they test lexicographically.

david-german-tri, 2016-04-26:
@liangfok I don't follow.  Are you saying component-based architecture requires tests to be in a separate subdirectory?  Why would that be true?

liangfok, 2016-04-26:
@david-german-tri: Looking up at @jwnimmer-tri's justification of his vote, he basically says:

> In many projects, I prefer the former (a separate test directory) to help keep filecount per directory lower. ... However, I've always had a fixed directory tree depth ...

Basically, he voted to keep the unit tests in the same directory as the code they test because of the high directory tree depth that's currently in Drake. I was responding saying I think Drake should adopt a more component-oriented architecture and a corresponding directory structure that is wider but flatter. If we're able to do that, I believe @jwnimmer-tri will change his vote to keep the unit tests separate.

Stylistically, I think unit tests are not part of the component and thus should be kept physically separated from the code that actually implements the component. Placing them in a subdirectory would achieve this.

amcastro-tri, 2016-04-26:
What about Matlab tests? should we have a `matlab_test` subdirectory?. 
I like @liangfok's component based view. In the future the Matlab code will only be wrappers to C++ code. I think then that these wrappers should be in a completely different directory tree separate from the C++ source tree. However now we have them all in one single place....

sherm1, 2016-04-26:
> Stylistically, I think unit tests are not part of the component and thus should be kept physically separated from the code that actually implements the component. Placing them in a subdirectory would achieve this.

This is the strongest point for me. I don't like the apples-and-oranges mix of test code with the component code.

I would like a flatter directory structure but I don't feel strongly that it should have maxdepth=2 if there is some *meaningful* substructure.

I think the "component architecture" discussion is confusing things here. I *think* @liangfok is referring to a set of independent modules connected by message passing, like the ROS architecture (please correct me if I'm wrong, Liang.) That's not on the table here! Drake builds tightly-coupled libraries (for good reason in places), so I think the rest of us are using "component" in a less specific way to indicate one of those libraries.

liangfok, 2016-04-26:
@sherm1: Yes, you are correct. Sorry for mucking up the discussion. It's just I feel adopting a more component-oriented approach will solve many of our problems, see, for example [this new conflict that I just discovered](https://github.com/RobotLocomotion/drake/issues/2187).

david-german-tri, 2016-04-27:
I have to say I don't get it (directories are not the same thing as build targets!), but it's pretty clear I am outvoted.  I'll update the style documentation.

liangfok, 2016-04-27:
INMHO, for conceptual clarity, the better the correlation between build targets and the physical co-location of files, the better.

david-german-tri, 2016-04-27:
I guess that's the fundamental disagreement.  I think it's common for source code to be conceptually related enough it should go in the same directory, but not mutually dependent such that it has to go in the same build target.  (Tests are one example, but this is also a common pattern in production code.  In Drake, I think we're likely to see it in `/common` especially.)

Anyway, not worth further resources to debate.

liangfok, 2016-04-27:
Agreed, this is becoming a philosophical / religious difference of opinion. :blush:

PR #2196 (2016-04-27): Add unit test style and location clarifications.
Closes #2182.  Codifies #2181.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/2196)
closed 2016-04-27 by david-german-tri