RobotLocomotion/drake #2187: Conflict Between ROS Message Generator and LCM Message Generator

liangfok, 2016-04-26:
When Drake's build system is [modified to generate custom ROS messages](http://wiki.ros.org/ROS/Tutorials/CreatingMsgAndSrv#Creating_a_msg), LCM messages are generated but the `[drake distro]/build/lib/python2.7/dist-packages/drake/__init__.py` file, which allows Python classes to _find_ the generated message `.py` files, is not correctly generated (it ends up being blank, see: https://github.com/RobotLocomotion/drake/issues/2179).

Specifically, adding the following line to `drake/CMakeLists.txt`, which is needed to generate custom ROS messages, results in the above-mentioned LCM message generation failure:

```
  # Generates the code for Drake's custom ROS messages.
  generate_messages(DEPENDENCIES std_msgs)
```

I'm going to label this issue "Kitware" since it is a build system issue.

liangfok, 2016-04-26:
Update: The problem is due to the ROS and LCM message generators both writing into the same directory (`[drake distro]/build/lib/python2.7/dist-packages/drake/`). I believe it's because both use the same CMake "project_name" when deciding where to put generated python files.

Not sure what's the best practice on how to solve this.

liangfok, 2016-04-26:
Commit https://github.com/liangfok/drake/commit/dc0d6c721d1ab847252202bb08b334df6bd369a7 moves ROS message generation before LCM message generation. This allows LCM messages to work but breaks ROS message Python generation. This is OK because we presently don't use custom ROS python messages. Since this is not a blocking issue, I will close it for now and re-open it once it does become a blocking issue.

closed 2016-04-26 by liangfok

jwnimmer-tri, 2016-04-26:
I am unconvinced that it will keep working under build parallelism (`-j8`).

liangfok, 2016-04-26:
:sob: