RobotLocomotion/drake #2190: Build failure: Error: Unable to find 'guile_scm.swg'

ErikSobel-TRI, 2016-04-26:
Tried to do a fresh clone/build on Mac OS X.  Build failed.
Making install in swig
:3: Error: Unable to find 'guile_scm.swg'
./nlopt.i:15: Error: Unable to find 'std_vector.i'
make[5]: *** [nlopt-guile.cpp] Error 1
make[4]: *** [install-recursive] Error 1
make[3]: *** [nlopt-prefix/src/nlopt-stamp/nlopt-build] Error 2
make[2]: *** [CMakeFiles/nlopt.dir/all] Error 2
make[1]: *** [all] Error 2
make: *** [all] Error 2

liangfok, 2016-04-26:
I haven't encountered this problem on my working OS X machine but here's what I see on my machine in case it helps to compare yours against mine:

``
$ cd [drake build artifacts directory]
$ find . -name "guile_scm.swg"
./build/share/swig/3.0.9/guile/guile_scm.swg
./externals/swig_matlab/swig-matlab/Lib/guile/guile_scm.swg
```

liangfok, 2016-04-26:
Have you tried to turn on the `DISABLE_MATLAB` option when you run `make options`?

ErikSobel-TRI, 2016-04-27:
Thanks for the suggestions:
1) Yes, I have the guile_scm.swg files in exactly the same location as you
2) Yes, I have turned ON the "DISABLE_MATLAB" flag

make options also says 
matlab :  not-found  
because there is no matlab installed on my system

But the error remains.

liangfok, 2016-04-27:
Not sure if this is feasible but could you try completely removing Mac Ports from your machine?

david-german-tri, 2016-05-19:
@ErikSobel-TRI is this still an open issue?

ErikSobel-TRI, 2016-05-19:
Nope. I solve that.

But I'm not 100% sure I remember what the solution was.  One thing we fixed
was the PYTHON_VERSION MAJOR MINOR not being specified correctly possibly
in drake/bindings/swig/CMakeLists.txt (it just specified python2 and not
python2.7)

Then there was a separate issue where cmake was picking up the wrong VTK
because I have built VTK-7.0 from source on my Mac as well.  But I think
that was a separate issue.

-Erik

closed 2016-05-19 by david-german-tri