RobotLocomotion/drake #2194: compareParsers.m Does Not Work On OS X (False Positive)

liangfok, 2016-04-27:
Unit test `[drake distro]/drake/systems/plants/test/compareParsers.m` does not work on OS X even though it is reporting a successful result. Here is the terminal output:

```
$ ctest -VV -R systems/plants/test/compareParsers
UpdateCTestConfiguration  from :/Users/liang/dev/drake/drake/pod-build/DartConfiguration.tcl
Parse Config file:/Users/liang/dev/drake/drake/pod-build/DartConfiguration.tcl
UpdateCTestConfiguration  from :/Users/liang/dev/drake/drake/pod-build/DartConfiguration.tcl
Parse Config file:/Users/liang/dev/drake/drake/pod-build/DartConfiguration.tcl
Test project /Users/liang/dev/drake/drake/pod-build
Constructing a list of tests
Done constructing a list of tests
Checking test dependency graph...
Checking test dependency graph end
test 147
    Start 147: systems/plants/test/compareParsers

147: Test command: /usr/local/Cellar/cmake/3.5.0/bin/cmake "-Dtest_name=systems/plants/test/compareParsers" "-Dadditional_paths=/Users/liang/dev/drake/build/matlab;/Users/liang/dev/drake/drake/systems/plants/test" "-Dtest_timeout=-1" "-Doutput_directory=/Users/liang/dev/drake/drake/pod-build/Matlab" "-Dworking_directory=/Users/liang/dev/drake/drake/systems/plants/test" "-DMatlab_PROGRAM=/Applications/MATLAB_R2016a.app/bin/matlab" "-Dno_unittest_framework=FALSE" "-DMatlab_ADDITIONNAL_STARTUP_OPTIONS=" "-Dunittest_file_to_run=" "-Dcustom_Matlab_test_command=try, eval('compareParsers'); catch ex, disp(getReport(ex,'extended')); disp(' '); force_close_system; exit(~strncmp(ex.identifier,'Drake:MissingDependency',23)); end; force_close_system; exit(0)" "-Dcmd_to_run_before_test=addpath_drake; global g_disable_visualizers; g_disable_visualizers=true" "-P" "/Users/liang/dev/drake/drake/cmake/MatlabTestsRedirect.cmake"
147: Test timeout computed to be: 1500
147: Matlab test  output:
147:
147:                             < M A T L A B (R) >
147:                   Copyright 1984-2016 The MathWorks, Inc.
147:                    R2016a (9.0.0.341360) 64-bit (maci64)
147:                              February 11, 2016
147:
147:
147: To get started, type one of these: helpwin, helpdesk, or demo.
147: For product information, visit www.mathworks.com.
147:
147:
147:    Home License -- for personal use only.  Not for government,
147:    academic, research, commercial, or other organizational use.
147:
147: Error using allURDFs (line 20)
147: still need to implement a version that works without find
147:
147: Error in compareParsers (line 8)
147:   urdfs = allURDFs();
147:
147:
147:
1/1 Test #147: systems/plants/test/compareParsers ...   Passed    8.61 sec

The following tests passed:
        systems/plants/test/compareParsers

100% tests passed, 0 tests failed out of 1

Total Test time (real) =   8.65 sec
```
I suspect CI is not catching this because `mac_matlab` column.

RussTedrake, 2016-05-07:
this works fine on my mac.

RussTedrake, 2016-05-07:
and it's running fine on the build servers:
linux: https://drake-cdash.csail.mit.edu/testDetails.php?test=605625&build=16785
mac: https://drake-cdash.csail.mit.edu/testDetails.php?test=604230&build=16821

@liangfok : i think you need to figure out why your machine is returning info~=1 for this line:
```
addpath_drake;
[info,p] = system(['find -L ',getDrakePath(),' -iname "*.urdf" | grep -v "/dev/" | grep -v "irb_140_convhull"']);
```

liangfok, 2016-05-10:
OK I will look into it. Thanks!

liangfok, 2016-05-14:
The method `allURDFs()` is defined in `[drake distro]/drake/util/allURDFs.m`. Here's the MATLAB code that appears to be failing on my machine:

```
if nargin<1, rootdir = getDrakePath(); end

urdfs = {};
[info,p] = system(['find -L ',rootdir,' -iname "*.urdf" | grep -v "/dev/" | grep -v "irb_140_convhull"']);
```

Using print statements, I verified that `getDrakePath()` is returning `/Users/liang/dev/drake/drake`.

I also verified that `find -L /Users/liang/dev/drake/drake -iname "*.urdf"` returns the paths to all URDF files within my `[drake distro]` directory. Unfortunately, once I pipe it to `grep -v "/dev/"` it results in no matches. I wonder if this is the problem.

liangfok, 2016-05-14:
The problem is I cloned `[drake-distro]` into `~/dev/`, and this causes `grep -v "/dev/"` to reject every URF in my `[drake-distro]` directory.

RussTedrake, 2016-05-14:
That's correct.  dev was a keyword in the drake directory structure saying "things that live beneath me are not subject to unit tests". You could make allURDFs more robust by removing only strongs which have dev after the drake root.

jwnimmer-tri, 2016-05-14:
It's straightforward to make the `find` correct, so that only dev subdirectories are ignored, not dev as a parent directory.  Use `find` with `-prune` for `-name dev` and the other blacklist item there; don't pipe the `find` through `grep` at all.

closed 2016-05-17 by jwnimmer-tri