RobotLocomotion/drake #2199: Windows buildslaves not coming up for experimental buidls

david-german-tri, 2016-04-27:
https://drake-jenkins.csail.mit.edu/job/experimental/1069/compiler=msvc-32,label=windows/console

It looks like the last time we actually got messages from a Windows build was 10:29:

https://drake-jenkins.csail.mit.edu/job/experimental/

@jamiesnape, any idea what's up?

jamiesnape, 2016-04-27:
Everything seems fine to me. There are eight Windows builds in progress, some of which started minutes ago.

sammy-tri, 2016-04-27:
https://drake-jenkins.csail.mit.edu/job/experimental/1061/compiler=msvc-32,label=windows/console

https://drake-jenkins.csail.mit.edu/job/experimental/1066/compiler=msvc-64,label=windows/console

https://drake-jenkins.csail.mit.edu/job/experimental/1067/compiler=msvc-32,label=windows/console

https://drake-jenkins.csail.mit.edu/job/experimental/1062/compiler=msvc-32,label=windows/console

https://drake-jenkins.csail.mit.edu/job/experimental/1062/compiler=msvc-64,label=windows/console

jamiesnape, 2016-04-27:
Looks like instance "drake-windows-2016-04-20-c4.4xlarge (i-406077db)" went bad. Not much I can do about that.

sammy-tri, 2016-04-27:
Does this mean that whatever happened with that instance is not expected to continue?

jamiesnape, 2016-04-27:
That instance is destroyed now, so yes.

david-german-tri, 2016-04-27:
Destroyed by manual action on your part, or some watchdog script, or just timing out?

jamiesnape, 2016-04-27:
Timing out. I am looking at ways of killing bad slaves earlier, but it is a work in progress.

david-german-tri, 2016-04-27:
If we wanted to kill them manually, is stopping them in the AWS UI good enough?

jamiesnape, 2016-04-27:
No, you need to delete them in Jenkins.

billhoffman, 2016-05-18:
Is this still an issue?

jamiesnape, 2016-05-18:
No. Bad instances are killed automatically.

closed 2016-05-18 by jamiesnape