RobotLocomotion/drake #2201: RigidBodyTree::inverseDynamics bug when acceleration is nonzero

tkoolen, 2016-04-27:
I recently looked over the RigidBodyTree::inverseDynamics code again and found that I had made a mistake that causes the torques returned by inverseDynamics to be wrong when `vd` (the desired acceleration vector) is nonzero. Luckily we've never used it with nonzero `vd` in the rest of the Drake codebase, but it still needs to be fixed of course. Just wanted to report it for now.

liangfok, 2016-04-27:
Thanks for reporting this. Just wondering what's the logic behind using the name "vd" for "acceleration vector"?

sherm1, 2016-04-27:
Wild guess: vd = time derivative of velocity?

tkoolen, 2016-04-27:
Yep.

amcastro-tri, 2016-04-28:
That's interesting that is never called with non-zero acceleration. What are the uses of this method then?

tkoolen, 2016-04-28:
Computing the Coriolis, gravity, and external force terms in the equations
of motion.

tkoolen, 2016-08-24:
I have the time to fix this now and write the corresponding tests. @sherm1, is that OK, or would you prefer to do it yourself? @liangfok, are you actively working on this code?

liangfok, 2016-08-24:
I am not working on this at this time. Would be great if you can fix this!

sherm1, 2016-08-24:
> I have the time to fix this now and write the corresponding tests. 

That would be great, Twan!

closed 2016-09-14 by jwnimmer-tri