RobotLocomotion/drake #2204: Update Drake Windows Installation Instructions with Visual Studio IDE Instructions

liangfok, 2016-04-28:
Right now, everything is command-line based. Add instructions on how to configure the Microsoft Visual Studio IDE to load the source code and compile it from within the IDE.

liangfok, 2016-05-17:
Here are some more comments from @sherm1:

> The first time through I have to click on `[build_dir]/drake-superbuild.sln` then do an `ALL_BUILD` in the VS GUI. When that finishes (a long time later!) I close that and click on `[build_dir]/drake/drake.sln` where I can make changes and rebuild frequently; that's where I see the project list I posted above. Setting PATH properly is required before that second step but I think that's already in the instructions. Currently both builds must be Debug, or both must be Release; I haven't been able successfully to switch between them on the fly (though that should be possible on Windows).

david-german-tri, 2016-10-06:
Windows is no longer supported.

closed 2016-10-06 by david-german-tri