RobotLocomotion/drake #2208: Simplify RigidBodyDepthSensor by Combining Pitch and Yaw Member Variables

liangfok, 2016-04-28:
The current [RigidBodyDepthSensor](https://github.com/RobotLocomotion/drake/blob/5878f08970311a3b24ea58fe32bfa8be1f97225b/drake/systems/plants/RigidBodySystem.h#L458) includes member variables for both pitch and yaw oriented laser scanners. However, only _one_ set of these parameters are used, see: [RigidBodyDepthSensor.cpp#L490](https://github.com/RobotLocomotion/drake/blob/5878f08970311a3b24ea58fe32bfa8be1f97225b/drake/systems/plants/RigidBodySystem.cpp#L490).

I recommend simplifying `RigidBodyDepthSensor` to not have separate "yaw" and "pitch" member variables:

```
  const std::shared_ptr<RigidBodyFrame> frame;
  double min_sense_angle;       // minimum angle of the camera FOV in radians
  double max_sense_angle;      // maximum angle of the camera FOV in radians
  size_t num_pixel;  // number of points in the image vertically (pitch)
  double min_range;       // minimum range of the sensor in meters
  double max_range;       // maximum range of the sensor in meters
```

The `frame` member variable should contain enough information to derive whether the sensor is scanning in the "yaw" or "pitch" directions.

A different sensor, perhaps called "PointCoudSensor", can be added that scans in both the yaw and pitch directions.

RussTedrake, 2016-04-28:
I think we're about to add kinect support and would prefer to keep it as is.

liangfok, 2016-04-28:
OK I suppose a 2D scanner can be thought of as a special type of 3D scanner in which the scanning range along one axis is zero.

I recommend changing the name of `RigidBodyDepthSensor` to be `RigidBodyPointCloudSensor`. This is because "Depth sensor" is singular and we're clearly measuring more than a single point.

Thoughts?

RussTedrake, 2016-04-28:
I don't see a big problem with depth, and point cloud could imply something different. (Not necessarily return raycast points). Plus this is the name in the matlab implementation.

liangfok, 2016-04-28:
Alright since this is arguably the simplest possible depth sensor and it can always be extended in the future, I will leave it alone. 

We will likely have to subclass it to add additional capabilities like support for color, multiple returns, no returns, and intensity values. But that will bring the subject of a future issue.

closed 2016-04-28 by liangfok