RobotLocomotion/drake #2211: Create a Drake System That's Able to Publish the Frame Transformation for ROS's tf Infrastructure

liangfok, 2016-04-28:
See: http://wiki.ros.org/tf

This is necessary to properly view the LIDAR sensor data being published in https://github.com/RobotLocomotion/drake/pull/2126 using  [RViz](http://wiki.ros.org/rviz) and other ROS GUI tools.

It is also needed to view the other models like the car and garage in said GUI tools.

liangfok, 2016-04-30:
I believe the appropriate component-based system architecture should be:

```
[Drake] →[Robot State Publisher] → [RViz]
```

References:
* http://wiki.ros.org/robot_state_publisher
* http://wiki.ros.org/rviz

liangfok, 2016-05-02:
I completed the initial `Drake` → `RViz` integration in https://github.com/liangfok/drake/commit/d5cc7a10062d2f53d48af8dc2008cc158fc3fb3c.

The `tf` transforms are being directly published by a Drake system, which are used by `RViz` to visualize the Prius. In the screenshot below, the left GUI is `Drake Visualizer` and the right GUI is `RViz`.

[image]

The next step is to visualize the LIDAR sensor data in RViz.

liangfok, 2016-05-03:
I completed the `Drake` + `Prius` + `4 LIDAR sensors` → `RViz` integration in https://github.com/liangfok/drake/commit/51c95802a4b9ac1fb87c4ffaa0f3e0e9c77270e9.

We are now able to visualize the results of the four LIDAR sensors mounted on the Prius in RViz:

[image]

Here's a video showing how the LIDAR sensor data changes as the car drives around in the Stata garage:

[image]

The next step is to clean up the code and then issue a PR.

jwnimmer-tri, 2016-12-22:
Done.

closed 2016-12-22 by jwnimmer-tri