RobotLocomotion/drake #2216: Polynomial does not aggregate terms

ggould-tri, 2016-04-29:
See the disabled test at https://github.com/RobotLocomotion/drake/blob/master/drake/util/test/testPolynomial.cpp#L232 .

The Polynomial `xy + xy` will not reduce to `2xy`.

This is also true of the commutative case (`xy + yx`); there is an argument to be made that this is correct (x and y could be of noncommutative type) but that goes against the unification of terms within a monomial (`yxy` -> `xy^2`) that Polynomial currently does.

This is not yet a crucial bug -- most current users of Polynomial don't care -- but will eventually result in missed opportunities for lumped variable identification.

Whoever fixes this bug should un-disable the corresponding test.

ggould-tri, 2016-05-25:
High priority as this directly impacts current work.  Assigning to myself because I may be the current expert on this code.

hongkai-dai, 2018-10-09:
I believe the current `symbolic::Polynomial` class will aggregate the terms, and also `xy` equals to `yx` in the `symbolic::Polynomial`. @ggould-tri OK to close this?

PR #10362 (2019-01-09): common: Enable polynomial_test case
At some point, this must have been repaired to be correct, and we didn't notice.

Closes #2216.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/10362)
closed 2019-01-09 by jwnimmer-tri