RobotLocomotion/drake #2217: RigidBodyTree and Dynamics Engine Needs to Support the Run-Time Addition and Removal of Branches in the Rigid Body Tree

liangfok, 2016-04-29:
This will become necessary once we're simulating huge worlds where the whole world can't be loaded simultaneously. Instead, only the parts of the world that are within the current region of interest should be loaded into the rigid body tree. These parts will, of course, change as the region of interest shifts over time.

Based on a recent discussion with @ErikSobel-TRI.

sherm1, 2016-04-29:
I agree that we'll want to add and remove things from a running simulation. But note: that does not necessarily require that we can *edit* a RigidBodyTree. Rebuilding it with changed contents is a practical alternative. Either way, transferring the still-valid portion of the existing state (context), and properly reinitializing the new state, is likely to require some careful thought.

sherm1, 2016-04-29:
For subsystems that are physically uncoupled we may just want to have multiple RigidBodySystems running, with whole systems being added and removed as needed.

liangfok, 2016-04-30:
Maybe we should look at how http://modlabupenn.org/ solves the problem, since they are obviously able to change the system over time with their robot modules.

liangfok, 2016-04-30:
I wonder if we could keep everything in the rigid body tree but not update or consider the parts of the tree that are currently outside of the current ROI (Region of Interest). To prevent discontinuities in the relevant parts of the model and thus controller output, we could begin updating these parts of the rigid body tree with the current state just-in-time before they enter the ROI.

jwnimmer-tri, 2018-12-21:
This issue is about the RBT code, in the attic. We're not going to fix it, it's okay to leave it however it is now.

There are already a couple other MBP-related issues similar to this; if they don't exactly capture this, we can open a new one, but this old one isn't helping.

closed 2018-12-21 by jwnimmer-tri