RobotLocomotion/drake #2218: Pendulum not stabilizable, traced to a diff in FeedbackSystem.m

peteflorence, 2016-04-29:
With current `master`, the policy from examples/Pendulum/runValueIteration.m will not be able to stabilize the pendulum in the simulation

Switching to the Dec 5 commit for systems/FeedbackSystem.m will make the pendulum stabilizable

There are only three commits since Dec 5, all on April 3: https://github.com/RobotLocomotion/drake/commits/master/drake/systems/FeedbackSystem.m

I'm still looking into it...

peteflorence, 2016-04-29:
It is the first of the April 3 commits that changes from stabilizable to not stabilizable: https://github.com/RobotLocomotion/drake/commit/107a563ffabbc39337a078b81b65d298b08c3116

peteflorence, 2016-04-29:
Okay that's about as much as I could figure out for now.

Clearly seems related to https://github.com/RobotLocomotion/drake/issues/494 but I'm not sure about all the history behind the zero crossings issues

RussTedrake, 2016-04-30:
Interesting.  Thanks for catching (and tracing) it!  The only think that really changed here is the integration scheme — it will be falling back to the main simulink solvers now instead of the DrakeSystem implementations.