RobotLocomotion/drake #2225: Drake's Directory Structure Should Reflect Project Focus and Algorithm Categories

liangfok, 2016-04-30:
Drake's algorithms and data structures can be separated into the following general categories, as specified  [here](https://github.com/RobotLocomotion/drake/blob/fa5cbb18d703fe133ede5ea01f7a0a4953bc9c21/drake/doc/doxygen.h#L6):

1. Modeling Dynamical Systems
2. Rigid-Body Kinematics and Dynamics
3. Simulation
4. Analysis
5. Planning
6. Feedback Control Design
7. State Estimation
8. System Identification

To better match the above categorization and provide greater conceptual clarity about Drake's objectives, Drake's source code should be reorganized into several top-level directories, each focusing on a different category and compiled into an independent library. This will solve the deep-directory-structure problem that was mentioned in https://github.com/RobotLocomotion/drake/issues/2182#issuecomment-214585072, and will likely simplify the build system since the dependencies will be better tailored to the needs of a particular library. The reorganization will also make it easier for users to pick-and-choose only the parts of Drake that they want to use.

Here's an initial proposal of how the directories should be structured. Ideally, each directory should be its own repository (so only the parts that are useful need to be downloaded). However, lumping all of them into a single repository for now would be OK.

```
[drake-distro]/drake_analysis/
[drake-distro]/drake_control/
[drake-distro]/drake_external_addons/
[drake-distro]/drake_examples/
[drake-distro]/drake_dynamical_system_modeling/
[drake-distro]/drake_lcm_integration/
[drake-distro]/drake_models/
[drake-distro]/drake_rigid_body_system_modeling/
[drake-distro]/drake_simulation/
[drake-distro]/drake_planning/
[drake-distro]/drake_ros_integration/
[drake-distro]/drake_state_estimation/
[drake-distro]/drake_system_identification/
```

The above-listed directories are just a suggestion; I'm sure we're going to have to iterate over them to get this perfectly right.

sherm1, 2016-04-30:
There is a lot of code in Drake that is shared between applications so I'm not sure how well it would work to organize the code by app. The proposed structure doesn't distinguish the shared "drake fundamentals" from particular applications that use those (e.g. util and common, but also models, rigid body stuff).

liangfok, 2016-04-30:
I think the dependencies between code in the top-level directories can be made explicit through documentation. For example, we can state in writing that "If you want to use `drake_dynamical_system_modeling`, you must also use `drake_rigid_body_system_modeling`." Stating these dependencies in writing, I think, is better than conveying them through a deep directory structure.

liangfok, 2016-04-30:
The top-level directories are akin to ROS' packages. Each package [lists their dependencies in writing](http://wiki.ros.org/ROS/Tutorials/CreatingPackage#ROS.2BAC8-Tutorials.2BAC8-catkin.2BAC8-CreatingPackage.package_dependencies). Tools like [rosdep](http://wiki.ros.org/rosdep) are provided that automatically download the dependencies needed by a particular package.

jwnimmer-tri, 2016-05-01:
I do think the physical structure of Drake's code can be improved, but I don't think those 8 items are the proper physical layering.  Instead, you'd want to look at coupling and cohesion, the current dependency graph of all of the existing C++ classes (and/or matlab functions), the programming languages being used, the stability of the APIs, which platform configurations they are supported on, and probably a few more factors that I'm forgetting at the moment.

Improving physical design is (to me) most successfully done through a more-bottom up approach (an inventory of the existing code, with place-holders for soon-to-be-forthcoming code), rather than a top-down approach based on the problem domain.

The logical design should likely be driven by the problem domain.  The physical design however, while closely related, often has a bunch of its own separate concerns.

liangfok, 2016-05-01:
I'm all eyes and ears. Looking forwarding to seeing and participating in the process of coming up with the best physical organization.

amcastro-tri, 2016-05-02:
I particularly agree with restructuring the directory structure. I can see how for instance there could be a "dynamics" directory with everything related to rigid bodies, joints and RBT's. Users could opt to only use the dynamics library without the need to link or include other packages like lcm or ROS if their application doesn't need them. 
I am not sure what the proper layers for other teams/functionalities would be but I can see how something like:
[drake-distro]/dynamics
[drake-distro]/math
[drake-distro]/math/optimization
[drake-distro]/math/linear-solvers
[drake-distro]/lcm
[drake-distro]/ROS-integration

could probably work.

david-german-tri, 2016-05-02:
I agree that a directory cleanup is needed, but I think it's too early to tell what the structure should be.  @jwnimmer-tri's points about C++ coupling and API stability are particularly critical, and particularly hard to assess right now.  I'd propose continuing with the current structure, with a tentative goal to clean it up in Q3 or Q4.

liangfok, 2016-05-02:
OK will close this issue now with the intent of re-opening it in 2016 Q3 or Q4.

closed 2016-05-02 by liangfok

liangfok, 2016-05-03:
Drake's current directory structure forces nested ROS packages, which is bad form. Specifically, it requires that `ROS_PACKAGE_PATH` include both `[drake distro]/drake` and example-specific packages within `[drake distro]/drake/examples/...` .