RobotLocomotion/drake #2228: completely remove matlab urdf/sdf parsers and dynamics implementations (in favor of c++)

RussTedrake, 2016-04-30:
this is a big ask, but clearly where we'd like to get to.  i'm opening the issue now so that we can collect the task list and so I can use this as an excuse to close some matlab parser issues that will eventually be resolved by this.
- [ ] the mex construct rbm interface will have to act more like addRobotFromURDF, and will pass the urdf/sdf file instead of passing the entire rigid body tree.
- [ ] will need to add additional mex entry points for adding links, joints, etc programmatically
- [ ] will need to verify that all of the use cases (trigpoly, etc) can get into and out of the c++ code successfully.
- [ ] will need to either add support for all of the rigid body sensors / actuators / forces in c++, or enable the c++ rigid body classes to callback into matlab.
- [ ] will get to zap a LOT of matlab code, and will only need to develop/improve the c++ code thereafter.

jwnimmer-tri, 2017-06-19:
Relates #6369.

jwnimmer-tri, 2017-07-25:
@RussTedrake I think perhaps this issue could be closed now?

closed 2017-07-25 by RussTedrake