RobotLocomotion/drake #2233: Superbuild externals should use shallow clone

jwnimmer-tri, 2016-05-02:
I think the superbuild externals should use a shallow git clone.  We don't typically want or need the full commit history of an external, and a build-from-scratch would go faster if all the git cloning were faster.

This forum post seems relevant, but I didn't immediately see any follow-up:
https://cmake.org/pipermail/cmake/2015-May/060571.html

bradking, 2016-05-31:
For reference, CMake 3.6 will add support for ExternalProject to do a shallow-clone.  See CMake commit [7db9f3f2](https://cmake.org/gitweb?p=cmake.git;a=commitdiff;h=7db9f3f294126e5e02d172e54c83a8ac90ed23a3).

bradking, 2016-05-31:
Shallow clones will not work for Drake's use case.  A shallow clone is useful when one wants a tagged release or the head of a branch.  It does not work when using an arbitrary fixed commit named by SHA-1.  The reason is that Git's fetch protocols only support fetching by ref, not by commit (for security reasons).  Therefore one must fetch one of the branch heads or tags that contains a given commit in order to get that commit.  If the desired commit does not happen to be exactly named by one of these refs then using `--depth 1` will fail to fetch the desired commit.  In general the needed depth is not known and changes over time for moving refs.

david-german-tri, 2016-06-08:
So, it sounds like we can fix this one-by-one for externals that have tags.  Probably shouldn't start until after #2487.

david-german-tri, 2017-01-13:
Closing this in favor of moving to Bazel.

closed 2017-01-13 by david-german-tri