RobotLocomotion/drake #2242: Support 3-component scale attribute in URDF/SDF

patmarion, 2016-05-02:
Currently, drake parses the scale attribute as a scalar value when it could be a 3 element vector.  There are example files in drake that use both scalar and vector attributes, perhaps they should all use vector attributes.

I developed a branch to resolve this last year.  I can't remember why the branch didn't get merged way back then.  It's just a single commit here:

https://github.com/patmarion/drake/commit/acbbacdbf53c20d47cda6abcf8b4c575fd34ff19

I have assigning this issue to @liangfok because I noticed that he changed the Prius model from scale="1" to scale="1.0 1.0 1.0" :)

patmarion, 2016-05-02:
Note, drake-visualizer already supports scale data as a scalar or 3 element vector.  As soon as BotVisualizer is updated to publish the data the visualization will work.

liangfok, 2016-05-02:
Yeah, let's make Drake standardize on a 3-vector for scale specification. If I recall correctly, I changed the scale in `prius.urdf` to make it compatible with ROS and Gazebo. Furthermore, the example on [this page](http://wiki.ros.org/urdf/XML/visual#line-57) also shows that scale is a 3-vector.

closed 2016-06-02 by jwnimmer-tri