RobotLocomotion/drake #2244: Problem visualizing large scenes

ErikSobel-TRI, 2016-05-02:
Users may have difficulty visualizing large scenes that use real world coordinates such as UTM.  This is a limitation of the VTK software used by director and drake_visualizer.  Scene files that contain large numbers such as an Wavefront (.obj) file with vertices like...
v  759186.1875 49.3500 -2957195.2500
v  759166.7500 73.1001 -2957193.2500
v  759167.0625 73.1001 -2957190.2500

Are either not read correctly or cannot be rendered correctly.  Structures will either not be present in the scene or rendered surfaces will appear and disappear randomly as the camera moves.  A short-term workaround is to subtract off a local origin from all coordinate for the purposes of rendering the scene.
Ideally the vtk data Reader would throw a warning or error when reading numbers that are too large to render.

liangfok, 2016-05-03:
@jwnimmer-tri and @bmartin-tri, this problem implies to me that there needs to be a decoupling between Earth's coordinate frame and the robot's coordinate frame. Since such an indirection is necessary, doesn't this impact our discussion about whether the X, Y, and Z axes of the robot's coordinate frame should conform with, for example, the coordinate frame used by a compass?

bmartin-tri, 2016-05-03:
@liangfok I don't follow.  Erik proposed a solution that was common at Jaybridge, where a local UTM origin was subtracted off of all coordinates before expressing them in simulator-local space.  I don't see what that has to do with rotating the compass heading.

liangfok, 2016-05-03:
@bmartin-tri, my point is since a transformation is necessary to convert between the robot's coordinate frame and the world's coordinate frame, the robot is free to operate in a coordinate frame that is completely different from the world's coordinate frame. Whether the transformation includes just a translation or translation + orientation makes little difference. Thus, the user of the robot should be free to select the coordinate frame that is most convenient for the particular application.

jwnimmer-tri, 2016-05-03:
I agree with @bmartin-tri.  The topics may be tangentially related, but are not really on point for this ticket.

patmarion, 2016-05-04:
VTK has support for double precision coordinate representations.  Unfortunately many of it's readers, especially the OBJ reader, are old and not great implementations, and will use floats that lose precision of the files.  In past projects using UTM coordinates I have always pre-processed data to subtract an origin, and stored the origin as metadata.

ErikSobel-TRI, 2016-05-25:
A valid workaround for this issue is to pre-process the data files, re-centering the scene coordinates so that they can be used with the appropriate data reader such as vtkOBJReader.

closed 2016-05-25 by ErikSobel-TRI