RobotLocomotion/drake #2250: LCM Message Modifications Not Applied Automatically By Build System

liangfok, 2016-05-03:
When modifying an LCM message definition, the newly generated python bindings are not automatically installed when issuing the `make` command. Two current work-arounds include deleting `[drake distro]/build/lib/python7.2` or executing `make install`. The build system should automatically detect changes in the LCM message definitions and install the newly generated python code without the user having to go through the above-mentioned workarounds.

mwoehlke-kitware, 2016-05-03:
Are you running `make` for the entire drake "superbuild", or just the subdirectory containing only drake itself?

The latter will not *and should not* update anything in the install prefix; that's the job of the `install` target. (If you're building just drake itself, it's probably a good idea to just always do a `make install`.)

jwnimmer-tri, 2016-05-03:
Ah, in my case I was running `make` in `drake-distro/drake`.

Perhaps the actual problem is that the python programs that are using the lcm messages are preferring to load from the install prefix, instead of the local drake build.

jwnimmer-tri, 2016-05-03:
(In other words, https://github.com/RobotLocomotion/drake/blob/master/drake/examples/Cars/README.md instructions to set PYTHONPATH to point at the superbuild is a lie.)

mwoehlke-kitware, 2016-05-03:
> Perhaps the actual problem is that the python programs that are using the lcm messages are preferring to load from the install prefix, instead of the local drake build.

I would guess that historically it has not been intended to support the case of an uninstalled drake newer than an installed drake. Or, possibly, to support an uninstalled drake at all. In particular, I'll note that the generated Python files aren't put somewhere that particularly looks like it's meant to be pointed at by `$PYTHONPATH`.

That said, I've seen projects that do try to support loading Python packages from a build directory. If we want to support an uninstalled drake, that's certainly something we could look at doing.

jwnimmer-tri, 2016-05-03:
Actually `drake-distro/drake/pod-build/lcmgen/lcmtypes` looks like a valid `PYTHONPATH` candidate to me.  What am I missing?

liangfok, 2016-05-03:
Aren't there many more scripts in `[drake-distro]/build/lib/python2.7/site-packages` that are needed?

```
$ cd [drake-distro]/build/lib/python2.7/dist-packages/
$ ls
bot_core  bot_lcmgl  bot_log2mat  bot_param  bot_procman  director  drake  pydrake  urdf_parser_py
```
Or, are you suggesting to add both `[drake-distro]/build/lib/python2.7/site-packages` and `drake-distro/drake/pod-build/lcmgen/lcmtypes` into the path?

mwoehlke-kitware, 2016-05-03:
> Actually `drake-distro/drake/pod-build/lcmgen/lcmtypes` looks like a valid `PYTHONPATH` candidate to me. What am I missing?

I said "particularly looks like it's *meant* to be pointed at", not "*could* be pointed at" :smile:. It does look like it may work to point `$PYTHONPATH` at `drake-distro/drake/pod-build/lcmgen` (note: the path should point at the *package(s)*, not the `.py` files *in* the package), but it's not obvious if that's by design (as e.g. if the path were `<build>/lib/python2.7/site-packages`; and yes, I've seen that in some projects) or just a happy accident. (Which makes me wonder if there are other hidden gotchas trying to use a not-installed drake.)

Mind, if someone *does* know that it's intended to work, then you can safely ignore me. Or, if we as a group decide we want to support this case, I expect we can fix any issues that come up. I'm just trying to avoid saying "yes, that definitely works" when it might not, or "yes, we should definitely support that" when the group as a whole might feel otherwise.

With that disclaimer... I'd say go ahead and try `drake-distro/drake/pod-build/lcmgen`. As @liangfok notes, you probably need that *in addition* to the other paths (which likely have important things from not-drake).

david-german-tri, 2017-01-03:
I'm going to close this as wontfix because it doesn't seem to be an actual pain point for anyone.  (No comments since May 2016.)

closed 2017-01-03 by david-german-tri

jwnimmer-tri, 2017-01-03:
I agree with closing it, because Bazel doesn't have this problem.  FYI its still a terrible hazard in CMake land, where there are multiple copies of built artifacts and you can easily end up using the wrong one (unless you are `drake/automotive/drake_paths.py` which carefully sets up the paths).