RobotLocomotion/drake #2252: Use swig3 whether MATLAB is present or not

david-german-tri, 2016-05-03:
the `WITH_SWIG_MATLAB` option loads the swig_matlab project which is based on swig3.  We should use swig3 even when we are just swigging Python, because it has significantly better C++11 support.

http://www.swig.org/Doc3.0/CPlusPlus11.html

jwnimmer-tri, 2016-05-03:
I believe that the superbuild `CMakeLists` actually has `WITH_SWIG_MATLAB` default to `ON` (even if matlab is absent), but CI purposefully sets `WITH_SWIG_MATLAB=OFF` for the non-matlab build configurations.  So this is just a specific instance of #2148.

https://github.com/RobotLocomotion/drake-ci/blob/58ad22b1a31d10e59898bd2bef2f0d971f1518e4/ctest_driver_script.cmake#L463

jamiesnape, 2016-05-03:
I can change that now, if you like.

david-german-tri, 2016-05-05:
Yes, please do!

david-german-tri, 2016-05-17:
https://github.com/RobotLocomotion/drake-ci/commit/2ad32bd2ec2db9ffec66f08407b75011943b33c0

closed 2016-05-17 by david-german-tri