RobotLocomotion/drake #2256: Drake System that is Compatible with ROS Joy is Needed

liangfok, 2016-05-04:
We need a Drake system that can take the steering / throttle commands from [ROS Joy](http://wiki.ros.org/joy) and pass them to the controllers in the car simulation. Once we have this, we can perform a close-loop integration with autonomous vehicle A.I. software.

liangfok, 2016-05-05:
This issue was resolved in commit https://github.com/liangfok/drake/commit/7dbd661dd17e3097f2cea453eea414f90a1297fd.

Here's a video:

[image]

I'll clean up the code and submit a PR soon. However, PR  https://github.com/RobotLocomotion/drake/pull/2253 should probably get merged first since this builds upon that.

liangfok, 2016-06-04:
I am waiting for https://github.com/RobotLocomotion/drake/issues/2369 to be closed before proceeding to issue PRs containing Drake systems that interact with ROS.

jwnimmer-tri, 2017-01-13:
@liangfok I am guessing this should be "backlog" and not "high"?

liangfok, 2017-01-13:
I'll set it to "medium" since folks are asking for it. It was working in System 1.0 and is now waiting on being ported to System 2.0.

jwnimmer-tri, 2017-02-09:
Ditto per #1759 comment -- I still advocate for "backlog" here.

closed 2018-06-09 by jwnimmer-tri