RobotLocomotion/drake #2259: Unit test ros_test needs to be updated to use rostest

liangfok, 2016-05-04:
See: http://wiki.ros.org/rostest

This will avoid needing to manually start `roscore` before running the unit test.

liangfok, 2016-06-06:
@jamiesnape: I need to create some directories and files and modify some environment variables in Drake-Linux-ROS-CI. I'm writing to you because your comment in https://github.com/RobotLocomotion/drake/issues/2369#issuecomment-220453697 indicates you may know how to do this.

Basically, I need to either run the following script, or replicate what this is script is doing, to convert the existing [ros_test](https://github.com/RobotLocomotion/drake/blob/master/drake/ros/test/ros_test.cc) unit test into an actual [rostest](http://wiki.ros.org/rostest):

https://github.com/liangfok/drake/blob/feature/issue_2259_ros_test/drake/ros/scripts/install_drake_ros_package.sh

Once the above changes are made, I need to modify Drake's unit testing framework to effectively execute the following command to run the `ros_test` unit test:

```
$ rostest drake_ros_package ros_test.test
```

Can you provide guidance on how the above can be achieved?

Thanks for any advice!

jamiesnape, 2016-06-07:
Are you using the `add_rostest` and/or `add_rostest_gtest` CMake functions?

liangfok, 2016-06-07:
I am not because Drake is not a real Catkin package using the Catkin build system. I am instead faking a Catkin package and system in a temporary directory to get `rostest` to work with Drake's ROS-based unit test.

jamiesnape, 2016-06-07:
I think we need to write some CMake functions that approximate `add_rostest`. The environment would be setup within that function. I will give that some thought and get back to you.

jhoare, 2016-06-28:
@liangfok I am looking at this now (faking a catkin workspace with our build tools), but as I am looking at it I am wondering - Why do you not want the ROS capabilities in its own catkin package, and build and use the ROS/catkin ecosystem? It would make the build a two step process: 1: build drake 2: build drake_ros_package 

How are you exporting what you've built to your ROS users? Are you just faking a built catkin workspace, and having them set the appropriate environment variables to see them?

liangfok, 2016-06-28:
> Why do you not want the ROS capabilities in its own catkin package, and build and use the ROS/catkin ecosystem? It would make the build a two step process: 1: build drake 2: build drake_ros_package

The main reason was I thought it would be harder to configure and build a ROS package that has Drake as an external. I could be wrong and would be delighted if doing that is not hard.

> How are you exporting what you've built to your ROS users? Are you just faking a built catkin workspace, and having them set the appropriate environment variables to see them?

I'm currently just exporting an exectable, see the subsection called "Run the Simulation" [here](https://github.com/liangfok/drake/blob/feature/ros_clean_exit/drake/examples/Cars/README_Drake_ROS_Car_Sim.md#run-the-simulation). The source code for that Drake + ROS integration is [here](https://github.com/liangfok/drake/blob/feature/ros_clean_exit/drake/examples/Cars/car_sim_lcm_and_ros.cc). It relies on several Drake systems defined [here](https://github.com/liangfok/drake/tree/feature/ros_clean_exit/drake/ros).

jhoare, 2016-06-28:
Alright thanks. I'm still very early on here, but I was able to get catkin to build the entire drake superbuild. I'll try to get those components from your branch also built, all within the same catkin workspace, but as their own ROS packages, and go from there.

liangfok, 2016-06-28:
STGM. I'm hoping your efforts will enable us to launch Drake via a ROS launch file, which is listed as a high priority item in https://github.com/RobotLocomotion/drake/issues/2606.

jhoare, 2016-06-28:
Made good progress, got everything building using catkin, although I did have to cheat a little. I'll try to clean it up and get it hosted and then we can talk about which way we want to go. 

I got your demo building & running and drove the prius around in the garage with my keyboard.

liangfok, 2016-06-28:
Awesome! I look forward to taking it your Catkin package for a test ride!

jhoare, 2016-06-29:
Created instructions here: 

https://gist.github.com/jhoare/59ea5aad4829bbd14490bdc4c4859f67

liangfok, 2016-06-29:
Thanks for writing the instructions! I was able to compile my Catkin workspace and run the car demo though there were a few issues:

**Issue 1:**

The command

 `git clone https://github.com/liangfok/ackermann-drive-teleop.git`

should be

`git clone https://github.com/liangfok/ackermann-drive-teleop.git ackermann_drive_teleop`

**Issue 2:**

Before running `catkin build`, I had to execute:

> `source /opt/ros/indigo/bash.setup`

**Problem running unit test:**

I tried to run the `ros_test` unit test but encountered the following error:

```
$ rostest drake_ros ros_test.test 
... logging to /home/liang/.ros/log/rostest-liang-MacBookPro-19137.log
[ROSUNIT] Outputting test results to /home/liang/.ros/test_results/drake_ros/rostest-test_ros_test.xml
testros_test ... FAILURE!
FAILURE: Test node [drake_ros/ros_test] does not exist or is not executable
  File "/usr/lib/python2.7/unittest/case.py", line 331, in run
    testMethod()
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rostest/runner.py", line 93, in fn
    self.fail(message)
  File "/usr/lib/python2.7/unittest/case.py", line 412, in fail
    raise self.failureException(msg)
--------------------------------------------------------------------------------

[ROSTEST]-----------------------------------------------------------------------

[testros_test][failed]

SUMMARY
 * RESULT: FAIL
 * TESTS: 0
 * ERRORS: 0
 * FAILURES: 1

ERROR: The following tests failed to run:
 * testros_test

rostest log file is in /home/liang/.ros/log/rostest-liang-MacBookPro-19137.log
```
Looking in my workspace, it seems as if the unit test wasn't building:

```
$ cd drake_ws
$ find . -name "*ros_test*"
./build/drake_ros/CMakeFiles/run_tests_drake_ros_gtest_ros_test.dir
./build/drake_ros/CMakeFiles/_run_tests_drake_ros_gtest_ros_test.dir
./build/drake_ros/CMakeFiles/ros_test.dir
./src/drake_ros/test/ros_test.test
./src/drake_ros/test/ros_test.cc
```

Do you know what could be wrong?

jhoare, 2016-06-29:
Not sure, I'll look into it and get back to you.

jhoare, 2016-06-30:
Had to change the way the tests were built, see this commit: https://github.com/jhoare/drake_ros/commit/e79e82faa873fea7fae6899cee2b69a3fe5df4ac

Should work for you now.

liangfok, 2016-06-30:
Thanks. Will test it soon.

liangfok, 2016-06-30:
I just pulled the latest from drake_ros and re-built the workspace but the error persists (ros_test.cc) is not being build into an executable.

In an attempt to debug this, I modified `drake_ws/src/drake_ros/CMakeLists.txt` to contain the following print statements and verified that both print statements showed up, so I'm still not sure what's wrong.

```
MESSAGE("*************in drake_ros CMakeLists.txt ****************************")

if (CATKIN_ENABLE_TESTING)
  MESSAGE("************* Adding ROS test!!! ****************************")
  find_package(rostest REQUIRED)
  include_directories(${CMAKE_INSTALL_PREFIX}/include)
  add_rostest_gtest(ros_test test/ros_test.test test/ros_test.cc)
  target_link_libraries(ros_test ${catkin_LIBRARIES})
  #add_test(NAME ros_test COMMAND ros_test)
endif()
```

Here's what I am executing to verify that no `ros_test` executable is being generated:
```
$ cd drake_ws
$ find . -name "*ros_test*"
./build/drake_ros/CMakeFiles/run_tests_drake_ros_gtest_ros_test.dir
./build/drake_ros/CMakeFiles/_run_tests_drake_ros_rostest_test_ros_test.test.dir
./build/drake_ros/CMakeFiles/_run_tests_drake_ros_gtest_ros_test.dir
./build/drake_ros/CMakeFiles/ros_test.dir
./build/drake_ros/CMakeFiles/run_tests_drake_ros_rostest_test_ros_test.test.dir
./src/drake_ros/test/ros_test.test
./src/drake_ros/test/ros_test.cc
```

jhoare, 2016-06-30:
Try running the following and see if its still not being built: 

``catkin build --verbose --no-deps drake_ros --make-args run_tests``

liangfok, 2016-06-30:
That worked! Any ideas why just running `catkin build` does not?

jhoare, 2016-06-30:
My guess is trying to build/run the ``run_tests`` target turns on the ``CATKIN_ENABLE_TESTING`` and compiles the binary, but I haven't looked into it much.

If we decide to go this route we would have to see how to properly hook in the ROS test running into CDash. Should I open a new ticket for that discussion?

liangfok, 2016-06-30:
I believe https://github.com/RobotLocomotion/drake/issues/2116 would be a good place to discuss how this drake-within-a-catkin-workspace option can be integrated into the CI infrastructure.

jhoare, 2016-06-30:
Even before that I was thinking there needs to be a discussion/decision on going forward, if ROS stuff for drake is built as part of Drake's build (as it is now) or if it will be separated out into its own catkin package, as my example shows is possible. I believe there's pros & cons to both approaches.

liangfok, 2016-06-30:
I think separating the Drake+ROS code into a separate Catkin package would be preferable assuming we can still maintain CI coverage for it.

liangfok, 2016-07-09:
@jhoare, I renamed the package `drake_ros` to be `drake_ros_systems` and combined the various Drake/ROS packages to be in the following repository:

https://github.com/liangfok/drake_ros_integration

This is because I believe there is a benefit to limiting the number of repositories that need to be kept in sync.

@RussTedrake, would be supportive of hosting the above-mentioned `drake_ros_integration` repository on the [RobotLocomotion github organization](https://github.com/RobotLocomotion/)?

RussTedrake, 2016-07-10:
I just took a look at https://github.com/liangfok/drake_ros_integration .  Those features are so important to drake that I would recommend putting them directly into the main drake repo.

jamiesnape, 2016-07-11:
I would vote the same as @RussTedrake. I was starting to replicate that sort of layout under drake/ros in #2653.

liangfok, 2016-07-11:
OK I'm going to look into whether we can combine all of the packages in [drake_ros_integration](https://github.com/liangfok/drake_ros_integration) with the main Drake repository. The biggest advantage I can see of that is everything will then be in a single repository.

jhoare, 2016-07-11:
Having a separate repository for the ROS integration is a fairly standard practice. 

How would we integrate everything into one repository? I see 2 options:
  1. Build with Catkin: Put the contents of the ``drake-distro`` repository along-side those in ``drake_ros_integration`` so that you've effectively built a catkin workspace, and ignore the ROS ones when you're not using catkin.
  2. Build without Catkin: Place the ROS code in a subdirectory of the ``drake-distro`` repository, and have our cmake build script fake the output as if it was built by catkin. - This was the path we were originally going down. 

I was hoping that we would be able to put catkin packages within catkin packages, (e.g. place the drake_ros packages in a subfolder of the drake main package) but this does not seem to work.

liangfok, 2016-07-11:
I would like to do everything possible to avoid faking a Catkin file system because I believe anything we fake will be brittle against changes to Catkin. I'm currently preparing a PR that does option #1 (placing `drake-distro` alongside `drake_ros_integration`).

liangfok, 2016-07-11:
FTR, the reason why we need a Catkin file system (whether real or faked) is so we can run `rostest`-based unit tests. These unit tests differ from regular unit tests in that they are started by taking as input a `.test` XML file and automatically (1) spinning up `roscore` and associated its services, (2) initializing the ROS parameter server with any parameters listed in the `.test` file, and (3) starting one or more ROS nodes listed within the `.test` file. These types of unit tests are declared using the [add_rostest_gtest()](http://wiki.ros.org/rostest/Writing#Add_rostests_in_CMakeLists.txt) method in CMakeLists.txt.

Since option #1 listed above will be a huge change to the current drake repository's directory structure (every single file will be moved into a subdirectory of the repository's root), I recommend that we first take an incremental approach by first merging #2764 before deciding whether to pull everything into a single repository.

jhoare, 2016-07-11:
In addition to testing, anyone who wants to use drake nodes from within the ROS system system (e.g. roslaunch files, ``rosrun pkgname nodename`` etc) would need at least the installed binaries/libraries and other build artifacts to resemble that of a ROS package.

jhoare, 2016-07-11:
An Option 3 would be to place all of the ROS code into the drake repository (e.g in ``drake/drake/ros``) and then add to the catkin/ros instructions to create a symbolic link into that directory, so catkin sees those ROS packages. 

The build steps would be along these lines:
```
$ mkdir -p ~/dev/drake_catkin_workspace/src
$ cd ~/dev/drake_catkin_workspace/src
$ git clone git@github.com:RobotLocomotion/drake.git
$ ln -s drake/drake/ros drake_ros_integration
$ cd ~/dev/drake_catkin_workspace
$ source /opt/ros/indigo/setup.bash
$ catkin init
$ catkin build
```

liangfok, 2016-07-11:
I like Option 3 since it enables us to keep everything in a single repository and yet allows Drake's ROS code to reside in a real Catkin package. I'll try it after #2764 gets merged. Thanks!

PR #2764 (2016-07-12): Updates to Drake to allow it to work with ROS.
This actually supersedes  #2763.

Closes #2259.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/2764)
closed 2016-07-12 by sherm1