RobotLocomotion/drake #2265: CI Should Verify PR Code Passes Clang-Format Unscathed

liangfok, 2016-05-04:
CI should run `clang-format` on a copy of each file in a PR and issue an error of the copies end up being different from the original files since that implies a style violation.

david-german-tri, 2016-05-04:
"CI should run clang-format on a copy of each file in a PR and issue an error if the copies end up being different from the original files since that implies a style violation."

This is not true.  `clang-format` often changes compliant code into other compliant code, and occasionally turns compliant code into an unreadable hash.  IMO, it's a useful tool but not ready for CI enforcement.

liangfok, 2016-05-04:
The problem is I, and I believe others, run `clang-format` without carefully studying everything it changes. Should we stipulate that any code that `clang-format` doesn't like be surrounded by 

```
// clang-format off
...
// clang-format on
```
comments?

david-german-tri, 2016-05-04:
I think that's a good idea for code that clang-format ruins (rare).  For code it transforms in trivial ways, I think `clang-format`-using authors such as yourself should feel free to make the change, but we don't need CI to force everyone else to do the same.

jwnimmer-tri, 2016-05-04:
I am very happy to put the responsibility of verifying clang-format propriety on the person who runs it. I don't think that developers should be allowed to blindly trust it.  They need to own every commit they make.

I not infrequently clang-format, decide that only ~75% of its fixes are worth committing, and then back out the unwanted 25%.  This is really easy in emacs (ediff-buffer).

Other times, I fix the code so that clang-format and I agree.  (Add a few parents, add a `// BR`, whatever.)  But mostly I am happy to just ignore it when it's flailing.

And for the record, if we do ever move clang-format up the ladder to being more required, we first need to spend some time tuning it.  There are a few knobs it has that would be better if they were tweaked.  We would also want to help people learn how to have their editor run clang-format for them while they type (most editors support this).

liangfok, 2016-05-04:
OK I'll close this issue and open https://github.com/RobotLocomotion/drake/issues/2266.

closed 2016-05-04 by liangfok