RobotLocomotion/drake #2269: Fix or suppress cpplint warnings for build/include_order

jwnimmer-tri, 2016-05-05:
We should fix cpplint's `build/include_order` warnings throughout our code (and/or suppress them on a one-by-one-basis), and then stop suppressing the warning globally in `drake-distro/drake/CPPLINT.cfg`.  This is a break-out from #2064.

Depending on what fixing the errors looks like, we might also decide to permanently suppress this warning globally.  I will follow-up later with more specific recommendation for this particular warning set.

jwnimmer-tri, 2016-05-05:
There was a discussion around clang-format and coding conventions that related to this, though I'm not immediately finding it in the logs.

I think the best approach for this issue is to teach clang-format how to properly sort our `#include` statements, and then apply it to solve this issue (probably only committing its changes to the include statements, not other random style fixes).

liangfok, 2016-05-05:
Could this be it? https://github.com/RobotLocomotion/drake/pull/1977

jwnimmer-tri, 2016-05-05:
Yup, that's the correct background discussion.  Thanks!

jwnimmer-tri, 2016-09-17:
In a new developer is familiar with clang-format configuration (or wants to learn), they could pick up this issue fairly easily.

jwnimmer-tri, 2017-02-27:
Updated proposal: we make a tool that calls `clang-format -lines 1..N somefile` with an `N` that covers the `#include` statements atop somefile.  (Let's call it `clang-format-includes` for now.) Users can run this tool (or teach their editor to run this tool) to format the include statements.  Then, we can either rely on cpplint to enforce it, or at test-time we could have the build enforce that running the tool is idempotent.

SeanCurtis-TRI, 2017-03-01:
There are instructions for CLion integration that gives you a *coarse* approximation of this.  You can simply select the lines with includes and do the "Clang Format Selection" external tool.  Not as nice as something automatic, but it does afford CLion users with the ability to order their includes without mangling the rest of the file.

jwnimmer-tri, 2017-03-01:
Awesome.  My main concern is that we have CI enforce it, and offer at least one tool (or docs) that automate it for all users.  We can cite that URL as part of this solution, too.

jwnimmer-tri, 2017-03-06:
On https://github.com/jwnimmer-tri/drake/tree/clang-format-includes, I wrote a `tools/clang-format-includes` program that formats `#include` statements for us, to be precisely what `cppguide` and `code_style_guide` ask for.  Even after running it, `cpplint` still has some complaints, because it doesn't understand which headers come from where.

Thus, the solution will be to leave this warning suppressed in `cpplint`, but to have `drakelint` enforce that `tools/clang-format-includes` is a no-op.

The PR train will start later this week.

jwnimmer-tri, 2017-08-01:
https://github.com/jwnimmer-tri/drake/tree/build-common-lint-pr5 has the final PR that will finish this -- include linting will be built-in to `bazel test ...`.

PR #6741 (2017-08-02): Add clang-format-includes to drakelint
Closes #2269.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/6741)
closed 2017-08-02 by jwnimmer-tri