RobotLocomotion/drake #2270: Fix or suppress cpplint warnings for build/include_what_you_use

jwnimmer-tri, 2016-05-05:
We should fix cpplint's `build/include_what_you_use` warnings throughout our code (and/or suppress them on a one-by-one-basis), and then stop suppressing the warning globally in `drake-distro/drake/CPPLINT.cfg`.  This is a break-out from #2064.

Depending on what fixing the errors looks like, we might also decide to permanently suppress this warning globally.  I will follow-up later with more specific recommendation for this particular warning set.

jwnimmer-tri, 2016-05-05:
These should be pretty easy to fix, and uncontroversial.  Very straightforward.

jamiesnape, 2016-05-10:
You can use the include-what-you-use job on Jenkins for this.

jwnimmer-tri, 2016-05-10:
Yes, though note that cpplint is only tagging missing libstdcxx includes, not the full suite of libraries that llvm can find in that build.

PR #3491 (2016-09-18): Remove build/include_what_you_use suppression
Fixes #2270.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/3491)
closed 2016-09-18 by jwnimmer-tri