RobotLocomotion/drake #2271: Fix or suppress cpplint warnings for build/namespaces

jwnimmer-tri, 2016-05-05:
We should fix cpplint's `build/namespaces` warnings throughout our code (and/or suppress them on a one-by-one-basis), and then stop suppressing the warning globally in `drake-distro/drake/CPPLINT.cfg`.  This is a break-out from #2064.

Depending on what fixing the errors looks like, we might also decide to permanently suppress this warning globally.  I will follow-up later with more specific recommendation for this particular warning set.

jwnimmer-tri, 2016-05-05:
These are super important and should be fixed with some priority, especially in header files.

jwnimmer-tri, 2016-09-17:
This would be a good onramp issue for a new developer.

I suggest something like this PR order:
- Start with a PR like #3490 -- kick the suppression into subdirs.
- Fix the warning in a directory at a time, in this order:
 - drake/common ASAP
 - drake/math ASAP
 - drake/solvers ASAP
 - any remaining header files that trigger this ASAP

Then with lower priority:
 - drake/systems as time permits
 - drake/examples is low priority
 - drake/util can stay suppressed
 - drake/matlab can stay suppressed

david-german-tri, 2016-09-20:
Heads-up for whomever works on this: at least some of the `using namespace std` in function scope are necessary to allow argument-dependent lookup of either an STL function or a custom function depending on template type.  

https://github.com/RobotLocomotion/drake/blob/master/drake/math/autodiff.h#L11

jwnimmer-tri, 2016-09-20:
Yes.  Don't just delete those, even if it appears to still compile.  Replace them with statements like `using std::sqrt` or whatever is referred to by the method.

david-german-tri, 2016-09-20:
> Replace them with statements like using `std::sqrt` or whatever is referred to by the method.

For reasons I don't understand, this doesn't work.  With https://github.com/david-german-tri/drake/commit/a03c8111d03607eff0c0643982ee1bee1b94219b, the compiler fails to find Drake's custom implementations of `floor` and `round`.

jwnimmer-tri, 2016-09-20:
Ah well.  For now they stay with a `// NOLINTNEXTLINE` I guess.

PR #5735 (2017-04-06): Fix remaining namespace lint.
Fixes #2271.  (Except for MATLAB stuff, which we'll ignore.)

@sammy-tri for all review.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/5735)
commit fbbd03cd (2017-04-06): Fix remaining namespace lint.

Fixes #2271.  (Except for MATLAB stuff, which we'll ignore.)
closed 2017-04-06 by sammy-tri