RobotLocomotion/drake #2274: Fix or suppress cpplint warnings for runtime/references

jwnimmer-tri, 2016-05-05:
We should fix cpplint's `build/include_order` warnings throughout our code (and/or suppress them on a one-by-one-basis), and then stop suppressing the warning globally in `drake-distro/drake/CPPLINT.cfg`.  This is a break-out from #2064.

Depending on what fixing the errors looks like, we might also decide to permanently suppress this warning globally.  I will follow-up later with more specific recommendation for this particular warning set.

jwnimmer-tri, 2016-05-05:
This is a tough one.  I know @sherm1 really wants to use non-const references as function parameters.  I don't think we should allow that (and neither does cpplint).

We should first decide whether we like the rule or not, and them update our cpplint integration to match that decision.

sherm1, 2016-05-05:
> I know @sherm1 really wants to use non-const references as function parameters

No, I'm willing to change my ways on that. I just never knew it was a problem before!

jwnimmer-tri, 2016-05-05:
Sounds promising!  (Though I don't have any immediate need to discuss; I'm happy to leave it to a PR-centric discussion or pre-discussion once it becomes relevant.)

jwnimmer-tri, 2016-09-17:
This is a good onramp candidate, once #3490 is merged.

PR #3955 (2016-10-31): List out specific runtime/references warnings in examples
The final portion of #2274. The prior related PRs are #3896, #3897, #3898, #3493.
We want to mark all of these now, so that they don't get any worse.

Closes #2274.  All of the remaining work is marked with TODOs in the source code, which should be fixed up as the code is otherwise reworked; there is no more use for an overarching issue.

---

This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/3955)
closed 2016-10-31 by david-german-tri