RobotLocomotion/drake #2287: Drake performance issues

amcastro-tri, 2016-05-06:
@sherm1 identified some possible performance bottlenecks in PR #2164 by @RussTedrake. Performance measurements where obtained with [Callgrind](http://valgrind.org/docs/manual/cl-manual.html) revealing that about 80% of the computational cost is spent within the SVD solver and in `BulletModel::closestPointsAllToAll`. In addition, the stability of the time stepping strategy forces the use of very small time steps. 

In summary they are:

1. Use Featherstone’s ABA instead of SVD
2. Increase time step stability by using a semi-explicit time stepping strategy
3. Improve on O(n^2) in `BulletModel::closestPointsAllToAll`

Item 3 will be addressed when solving #2004 (currently in progress).

RussTedrake, 2016-05-06:
Excellent.  How much time is in the actual svd solve?

sherm1, 2016-05-06:
> Excellent. How much time is in the actual svd solve?

It was 40% of the total time.

RussTedrake, 2016-05-06:
Eigen has a number of least squares solvers, it could definitely be worth looking into more of them.  Also - was it possible to calgrind in release mode?  Eigen performance will be very different in debug vs release.

sherm1, 2016-05-06:
> Also - was it possible to callgrind in release mode?

I did those measurements in a release build, specifically: `gcc 5.3.1 -O2`

tri-ltyyu, 2019-03-22:
This is likely specific to RigidBodyPlant. If this is applicable to MultBodyPlant/SceneGraph, please file new issue. Closing for now.

closed 2019-03-22 by tri-ltyyu