RobotLocomotion/drake #2290: Update "Tools for Code Style Compliance" on clang-format

amcastro-tri, 2016-05-06:
The documentation on [usage of clang-format](http://drake002.csail.mit.edu/drake/sphinx/code_style_tools.html) in the wiki needs to be updated on instructions on how to specify Google's style. Now it simply states to pass `-style=file` to clang-format.

liangfok, 2016-05-06:
The current instructions are:

```
$ clang-format -i -style=file [file name]
```

The `-style=file` tells `clang-format` to look for a [.clang-format](https://github.com/RobotLocomotion/drake/blob/master/.clang-format) file, which specifies the Google style. Doesn't this already specify the Google style?

liangfok, 2016-05-06:
Are you suggesting that the instructions should be updated to use the following instead?

```
$ clang-format -i -style=Google [file name]
```

jwnimmer-tri, 2016-05-06:
Ah, perhaps I misunderstood what was happening, in the code review that led to this ticket.  Perhaps the instructions are already correct.

sherm1, 2016-05-06:
I think the instructions are correct, but they depend on being in the right directory subtree -- my theory is that @amcastro-tri may have been running `clang-format` from a directory where no `.clang-format` file existed up the directory tree.

amcastro-tri, 2016-05-06:
I was running from the `[drake distro]` directory. Is that the cause?

jwnimmer-tri, 2016-05-06:
@amcastro-tri And what version of clang-format, and on what OS?

amcastro-tri, 2016-05-06:
Ok. Either from [drake distro] or [drake distro]/drake I get the same result. This is Ubuntu 14.04 and clang-format 3.7.1.
I get the same result if using `style=Google`.

liangfok, 2016-05-06:
So it works now? I'm tempted to change the instructions to use `-style=Google` if it is more robust against execution of `clang-format` from outside of the `[drake distro]` directory.

amcastro-tri, 2016-05-06:
not yet. What I am saying is that I get the wrong format from either [drake distro] or [drake distro]/drake with either `style=file` or `style=Google`.

jwnimmer-tri, 2016-05-06:
Interesting.  For me as well, `clang-format-3.6 -style=Google compareParsersmex.cpp` is putting the `*` in the wrong place on line 57 (after changing it from smart_ptr to raw pointer).  I think @sherm1's suggestion that @amcastro-tri was using the LLVM format by accident has perhaps misled us.

I think the actual problem is that `DerivePointerAlignment` is on in Google style, and the mex file we have here is full of wrong-code, so clang-format is keeping it consistent, rather than making it be what we want.

sherm1, 2016-05-06:
Sorry to add to the confusion! I just tried clang-format 3.7 and 3.8 on Ubuntu 14.04 & 16.04 on this program:
```c++
int main() {
  int* p = 0;
  return 0;
}
```
With style=LLVM it moves the `*` next to the p. With style=Google it leaves it alone. However, if I change to 
```c++
int main() {
  int *p = 0;
  return 0;
}
```
Then *both* styles leave it alone. So it looks like the Google style says "leave it as found". However, if I added a few more pointers like this:
```c++
int main() {
  int* a;
  int* b;
  double* c;
  int *p = 0;
  return 0;
}
```
Then LLVM moves every `*` to the right, while Google moves the last `*` to the left to make them consistent.

So the Google rule actually seems to be to leave things alone if they are consistent, otherwise make them consistent. I would prefer that we instruct clang-format to move `*` and `&` next to the type unconditionally.

jwnimmer-tri, 2016-05-06:
@sherm1 Yup!  Indeed, that is what #2291 does.

PR #2291 (2016-05-06): Tune up clang-format to force pointers to the type for C++
Tune up clang-format to force pointers to the type for C++, even if the predominant convention in the file is otherwise.  This closes #2290 (at least, the underlying confusion).

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/2291) &emsp; Multiple assignees:&emsp;[image]&nbsp;<a href="/robotlocomotion/drake/pulls/assigned/amcastro-tri">amcastro-tri</a>,&emsp;[image]&nbsp;<a href="/robotlocomotion/drake/pulls/assigned/david-german-tri">david-german-tri</a>
closed 2016-05-06 by jwnimmer-tri