RobotLocomotion/drake #2292: Some possible constantness issues identified.

amcastro-tri, 2016-05-08:
Some constantness issues were identified by @jwnimmer-tri in #2207. Resolving these issues in #2207 would make the PR even bigger of what it already is and therefore this new issue is opened. 

- `RigidBodyFrame::body`: RigidBodyFrame should know what body it belongs to but most likely it should not modify it. Therefore this member should be a constant pointer to a constant RigidBody. 
- `RigidBodyTree::findLink`: this method should probably return a constant pointer.

jwnimmer-tri, 2016-05-08:
To be clear, on things like findLink, the problem is a const method returning a non-const pointer into our representation.  If calling code wants to get non-const pointers out of us that's fine, but they shouldn't use an object-const method to do it.

liangfok, 2016-05-08:
Regarding:

> RigidBodyTree::findLink: this method should probably return a constant pointer.

Instead of "return a constant pointer" do you mean "return a pointer to a const member variable"? Or, do you mean "return a const pointer to a const member variable"?

closed 2017-02-01 by amcastro-tri