RobotLocomotion/drake #2293: port changeRootLink functionality to C++

manuelli, 2016-05-09:
currently the changeRootLink functionality #417 is only in c++. Now that we have Valkyrie we have pelvis and back mounts for testing. Since the pelvis is the root link we can only have a fixed pelvis model in Drake. For testing purposes it would be good to have a fixed back model as well.

Discussed this briefly with @RussTedrake @tkoolen. Assigning to myself for now but would like to hand this off.

liangfok, 2016-05-09:
As a quicker workaround, couldn't we simply modify the SDF / URDF to make the torso the parent of the pelvis rather than the other way around?

manuelli, 2016-05-09:
I'm not sure that is so easy because there are lots of offsets and rotations in the urdf specification that are specific to the ordering that is given.

RussTedrake, 2016-05-09:
Plus we want to keep compatibility with the urdfs used by the other Val teams.

RussTedrake, 2016-05-27:
for reference, the matlab implementation is here:
https://github.com/RobotLocomotion/drake/blob/master/drake/systems/plants/%40RigidBodyManipulator/changeRootLink.m

liangfok, 2016-05-27:
Sorry I'm still trying to understand. Isn't the "root link" of a robot defined by the link to which we attach the virtual floating joint? If so, couldn't we make the torso the root link by simply attach the floating joint to the torso rather than the pelvis?

tkoolen, 2016-05-27:
Unfortunately it's not quite that trivial. The problem is that there are various objects associated with rigid bodies, like visuals, collision objects, etc. The pose of these objects (both in the URDF and in our RigidBodyTree) is defined *with respect to the frame after the parent joint*. If you change the root link, then some parent joints may flip direction and become child joints. So you have to change the pose of all the objects that are associated with a rigid rigid body if its parent joint flips direction.

I think the long term solution should be to take all of the objects currently associated with rigid bodies and instead associate them with frames. Then you would only have to change the definition of that frame, instead of having to keep track of all of those objects and transform them individually. But again that's longer term and it would be good to get this soon.

sherm1, 2016-05-27:
Also longer term: IMO it is a shame to get the links and joints mixed up with their (non-unique) representation as an internal coordinate tree. Links and joints and the meaning of joint coordinates are all very user-relevant, while the internal model ought not to be. The same user-facing behavior is achievable with maximal coordinates or a variety of minimal coordinate representations.

tkoolen, 2016-05-27:
Yep, completely agree with that.

liangfok, 2016-05-27:
Thanks for the explanation and longer term solution! It makes sense.

liangfok, 2017-03-09:
I recommend that this feature be implemented in `MultiBodyTree` rather than `RigidBodyTree`.

\CC: @amcastro-tri

tri-ltyyu, 2019-03-22:
This is likely specific to RigidBodyPlant. If this is applicable to MultBodyPlant/SceneGraph, please file new issue. Closing for now.

closed 2019-03-22 by tri-ltyyu