RobotLocomotion/drake #2300: Why HybridTrajectory requires that state frame to be the same in different modes?

hongkai-dai, 2016-05-09:
In Line 28 of HybridTrajectory, it requires that the state frame to be the same in different modes. But this prevents us to construct a hybrid trajectory for SLIP, in which the flight phase has state frame `x,y,xdot,ydot`; while the stance phase has state frame `r,theta,rdot,thetadot,x_toe`

sherm1, 2016-05-09:
@david-german-tri, this is relevant for System 2.0 also. A discrete event (detected by a witnessed zero-crossing) invokes an event handler ("reset method") that changes "mode" (aka "phase"), and that can cause discrete changes not only to the values of the state variables but to their dimensions.

RussTedrake, 2016-05-10:
Yes — this was an artificial constraint due to the frame implementation in matlab.  it would be great to avoid it in System 2.0.

david-german-tri, 2016-05-10:
Thanks for the heads-up.  We can definitely accommodate this in System2.  (The more requirements like this we accrue, the less comfortable I am with the Context-is-a-monolith design, but that's a separate issue.)

RussTedrake, 2016-05-25:
@hongkai-dai -- my instinct is that this will not get fixed in matlab (unless you need it), but will be fixed in c++.  if you agree, then maybe you can close this?

hongkai-dai, 2016-05-25:
I do not need this feature, but people in UCLA had this problem when they run trajectory optimization for SLIP model (it has different number of states in each hybrid modes). I can let them know that we are not going to fix this problem in MATLAB.

closed 2016-05-25 by hongkai-dai