RobotLocomotion/drake #2301: AllBodiesClosestDistanceConstraintTest uses wrong active_collision_options

hongkai-dai, 2016-05-10:
@avalenzu It looks to me that at line 39 of AllBodiesClosestDistanceConstraintTest, 
```
con = AllBodiesClosestDistanceConstraint(r,0.02,1e3,[0,1]);
```
the 4th argument is not set correctly. It should be a struct representing `active_collision_options`, instead it passes `[0,1]`, that looks like the `tspan`. Could you take a look?

hongkai-dai, 2016-05-10:
Actually, the `active_collision_options` is not used at all in the `AllBodiesClosestDistanceConstraint`, is it OK if we just remove it?

david-german-tri, 2016-05-10:
@amcastro-tri, you might be interested in this also.

RussTedrake, 2016-05-25:
ok with me!

jwnimmer-tri, 2017-06-19:
Closing as won't-fix, because #6369 will be removing the relevant software.

If I'm confused and this should in fact stay open, please re-open an clarify the remaining unsolved problem(s) in a new comment here.

closed 2017-06-19 by jwnimmer-tri