RobotLocomotion/drake #2305: examples/Pendulum/runTrajectorySwingUpAndBalance is flaky

jamiesnape, 2016-05-10:
```
380/461 Test #379: examples/Pendulum/runTrajectorySwingUpAndBalance ...............................***Failed  138.52 sec
LCM: Disabling IPV6 support
LCM: TTL set to zero, traffic will not leave localhost.
Matlab test  output:

                            < M A T L A B (R) >
                  Copyright 1984-2015 The MathWorks, Inc.
                   R2015b (8.6.0.267246) 64-bit (maci64)
                              August 20, 2015

 
To get started, type one of these: helpwin, helpdesk, or demo.
For product information, visit www.mathworks.com.
 

	Academic License

LCM: Disabling IPV6 support
LCM: TTL set to zero, traffic will not leave localhost.
 Calling addpath_spotless
 Calling addpath_sedumi
 Calling addpath_mosek
Error using javaaddpathProtectGlobals (line 27)
Could not modify the Java path because there are already Java objects present. You should either run 'clear java' yourself or ensure that this function is called before any Java objects are created. If possible, move any calls to checkDependency to the beginning of their function or script.

Error in addpath_mosek>javaaddpathIfNew (line 16)
    javaaddpathProtectGlobals(p);

Error in addpath_mosek (line 6)
    javaaddpathIfNew(jarfiledir);

Error in checkDependency>pod_pkg_config (line 461)
      eval(cmd);

Error in checkDependency (line 215)
        conf.mosek_enabled = pod_pkg_config('mosek') && logical(exist('mosekopt','file'));

Error in PolynomialSystem/regionOfAttraction (line 49)
ok_mosek = checkDependency('mosek');

Error in PendulumPlant/balanceLQR (line 73)
        V=regionOfAttraction(pp,V,options);

Error in PendulumPlant/trajectorySwingUpAndBalance (line 116)
      [ti,Vf] = balanceLQR(obj);

Error in runTrajectorySwingUpAndBalance (line 7)
c = trajectorySwingUpAndBalance(pd);

 Calling addpath_snopt
Elapsed time is 3.877319 seconds.
Starting parallel pool (parpool) using the 'local' profile ... connected to 4 workers.

sedumi: infeasible mss program
Error using msspoly/subs (line 24)
Second and third argument must have same dimensions.

Error in mssprog/get (line 12)
a=subs(q,pr.v,pr.x);

Error in mssprog/subsref (line 7)
                y=get(pr,s.subs{1});

Error in PolynomialSystem/sampledFiniteTimeVerification>optimizeRho (line 480)
  rho = doubleSafe(prog(rho));

Error in PolynomialSystem/sampledFiniteTimeVerification (line 148)
      [rho,rhointegral]=optimizeRho(x,V,Vdot,L,dts,Vmin,rhof,options);       

Error in PendulumPlant/trajectorySwingUpAndBalance (line 126)
      Vswingup=sampledFiniteTimeVerification(psys,xtraj.getBreaks(),Vf,Vswingup,options);

Error in runTrajectorySwingUpAndBalance (line 7)
c = trajectorySwingUpAndBalance(pd);

 

CMake Error at /Users/yosemite/workspace/mac-clang-continuous-matlab/drake/cmake/MatlabTestsRedirect.cmake:108 (message):
  [MATLAB] TEST FAILED Matlab returned 1
```

RussTedrake, 2016-05-25:
@anirudhamajumdar -- any idea what this is failing sometimes?

RussTedrake, 2016-05-25:
(the test should be deterministic).  if failure is inevitable, can we make the code handle it more gracefully?

jwnimmer-tri, 2017-06-19:
Closing as won't-fix, because #6369 will be removing the relevant software.

If I'm confused and this should in fact stay open, please re-open an clarify the remaining unsolved problem(s) in a new comment here.

closed 2017-06-19 by jwnimmer-tri