RobotLocomotion/drake #2306: RigidBodyConstraint does not conform to style guide

liangfok, 2016-05-10:
Mostly it's just the locations of the `&` and `*` symbols.

liangfok, 2016-05-11:
Oops, I mistakenly assigned this to @amcastro-tri thinking he was working on this class. @amcastro-tri, if you are not working on this class, please re-assign it to someone who is.

sammy-tri, 2016-05-11:
I've been in here lately, I can take a swing at it.  @amcastro-tri, feel free to reassign to me if you like.

liangfok, 2016-05-11:
Thanks @sammy-tri!

amcastro-tri, 2016-05-11:
Thanks @sammy-tri!!

sammy-tri, 2016-05-12:
PR #2325 fixes some of the style guide conformance issues.

jwnimmer-tri, 2016-05-12:
For whoever next picks up this issue, please start by breaking up RigidBodyConstraint.h/cpp into separate, smaller files (with appropriate snake_case_name.cc).  That will make it easier to iterate on the remaining style fixes.

sammy-tri, 2016-05-12:
This shouldn't actually be closed yet.  #2325 wasn't nearly enough to fix all the style problems...

sammy-tri, 2016-05-16:
Removing myself as assignee -- I don't have any more cleanups planned in the near future.

sherm1, 2018-08-14:
I think it is safe to say we won't be going any further with this cleanup.

PR #2325 (2016-05-12): Fix some issues with RigidBodyConstraint not conforming to the style guide.
Fixes #2306...   well, partly.  I could, I think, spend days cleaning up in there.  But it fixes the spacing issue identified there along with access controls and variable names.  I did not get to function names, namespaces, etc...

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/2325)
closed 2018-08-14 by sherm1