RobotLocomotion/drake #2316: linkname Needs to be Updated to Conform To Style Guide

liangfok, 2016-05-11:
`RigidBodyTree::linkname` should become `RigidBodyTree::link_name_` and ideally should be private.

Parameters and local variables named `linkname` throughout the code need to become `link_name`.

The Matlab code needs to be updated to match the C++ code.

Since this touches to much code in Drake, I would consider this a major operation.

jwnimmer-tri, 2016-05-11:
You can change the C++ code, but leave the matlab API the same, via the `*.i` file wrapper specifications.

liangfok, 2016-05-11:
True, but would it be bad that the C++ and Matlab code diverge in variable names?

jwnimmer-tri, 2016-05-11:
It's a tradeoff.  To me, leaving matlab alone is probably worth the divergence.  It's essentially the public API of Drake right now.  We could provide both spellings to matlab if we wanted to, though perhaps that is even worse.

sherm1, 2016-12-22:
@amcastro-tri, I'm assigning this to you since you are modifying RigidBody.

amcastro-tri, 2016-12-26:
This is actually solved. @liangfok renamed it to a private member `RigidBody::name_` with public setter `set_name()` and getter `get_name()`.

closed 2016-12-26 by amcastro-tri