RobotLocomotion/drake #2323: RigidBodySystem::spatialForceInFrameToJointTorque() Needs to be Moved to RigidBodyTree.

liangfok, 2016-05-11:
See the following review comment for context:  

https://reviewable.io/reviews/robotlocomotion/drake/2303#-KHROOYoXMVoJcMHqdJA

liangfok, 2016-05-11:
Also consider renaming the method to be `CoputeGeneralizedJointForcesFromWrenchOnRigidBodyFrame()`.

See: https://reviewable.io/reviews/robotlocomotion/drake/2303#-KHRMPUfS_OpOwvf4itk

amcastro-tri, 2016-05-12:
Ok, I can take this one. So I assume your plan is then to leave `spatialForceInFrameToJointTorque` in `RigidBodySystem.h` as it was originally? will you revert that change in #2303?

liangfok, 2016-05-12:
I moved the implementation of `spatialForceInFrameToJointTorque` from the `.h` file to the `.cpp` file. Since I believe this is an improvement, would it be OK to keep that refactor in https://github.com/RobotLocomotion/drake/pull/2303? If not, I would be happy to revert that change as well.

amcastro-tri, 2016-05-12:
You mean because it would make the build faster? in that case I agree

liangfok, 2016-05-12:
Yes, to make the build faster and de-clutter the `.h` file. OK I will keep the refactor in https://github.com/RobotLocomotion/drake/pull/2303.

amcastro-tri, 2016-05-12:
sounds good.

sherm1, 2016-05-12:
Copied from reviewable for preservation:

@amcastro-tri:
> We need proper Doxygen documentation here. My try:
"Given a wrench described in the given RigidBodyFrame frame, return the (scalar) generalized force at each joint required to achieve this wrench".
@sherm1. What do you think?, because if a joint is prismatic (and if I understand correctly) the value of tau will be that of the (scalar) force along the axis of action. Therefore, the name of this method is misleading. It should contain the term "generalized force", not "torque".
My try: rename to `ComputeGeneralizedJointForcesFromWrenchOnRigidBodyFrame()`.

@sherm1:
> It looks like Drake is using the term "spatial force" rather than "wrench". Those are synonyms. Usually I would prefer the shorter term, but this terminology has the advantage that it can be used for other "spatial vector" (6 element) quantities like spatial velocity and spatial acceleration. We ought to pick a terminology and stick with it.

> Also, I would reword the proposed description a little. An applied spatial force (or wrench) produces a set of generalized forces at the joints that will produce the same motion. But generalized forces cannot be said to "achieve" that wrench; it just achieves the same acceleration. Also, a single joint may have several generalized forces (up to 6). So how about: "Given a spatial force (wrench) applied at the origin of the given frame, and expressed in that frame, return the vector of generalized forces induced by the spatial force."

> Could be `ConvertSpatialForceToGeneralizedForces()`. I'm not sure the "OnFrame" part is adding anything because a spatial force must inherently be applied to a point (for the force half), and must be expressed in a frame (since both force and torque components are vectors). The method is a little odd because it combines the application point and expressed-in frame. Usually spatial force are most naturally computed in the World frame and then applied to some point on a body.

amcastro-tri, 2016-05-12:
That's great! thank you @sherm1. It would've been difficult to find that again

jwnimmer-tri, 2018-09-27:
I assume this issue is WONTFIX.  Please reopen in case not.

closed 2018-09-27 by jwnimmer-tri