RobotLocomotion/drake #2324: Buildcop: Windows builds failing 2015-06-11 afternoon due to agent/connection issues

david-german-tri, 2016-05-11:
Example continuous failure: https://drake-jenkins.csail.mit.edu/view/Continuous/job/windows-msvc-32-continuous-release/216/console

Example experimental failure: https://drake-jenkins.csail.mit.edu/job/experimental/1361/compiler=msvc-64,label=windows/console

@amcastro-tri FYI

billhoffman, 2016-05-11:
@jamiesnape and I have been talking about this.  Is there a good place to send out alerts for stuff like this?  Seems like the project needs a drake-developers email list to announce stuff like this and changes in workflow/etc.

david-german-tri, 2016-05-11:
If there's an actual problem to be resolved, I think opening a GitHub issue serves that purpose.  For other types of announcements, I can't think of a good existing list.  As a short-term solution, you can email me and I'll at least disseminate the news within TRI; longer-term, perhaps we can get drake-developers@mit.edu.

In any case, what's up with this particular event?

billhoffman, 2016-05-11:
Not actually sure yet.  Things started going bad after @jamiesnape patched the git plugin that was meant to fix the missing git hashes on matrix builds.  He just rolled it back, and restarted the PR builds that failed.  Watching it now, seems to be working, but nothing has completed yet.

jamiesnape, 2016-05-11:
So far, not too much makes sense, but that is the other thing that changed. The Windows image changed slightly yesterday with the WinRM fix, but there were no problems overnight of note.

jamiesnape, 2016-05-12:
I rolled back the git patch and everything passed yesterday afternoon/evening. I reapplied the patch late evening and all the nightlies passed, so I honestly have no idea. Restarting the all the jobs yesterday after the outage was a pretty good stress test, so there is no obvious problem now.

jamiesnape, 2016-05-13:
Jenkins left nothing helpful in its logs and the console messages, if any, were pretty generic, so there is not much more we can do here. We had no further occurrences of the particular error yesterday and at one point in the afternoon had a record 46 builds across all platforms running simultaneously without issue. I am as puzzled as everyone else, but realistically, I suggest we close this issue.

closed 2016-05-13 by david-german-tri