RobotLocomotion/drake #2326: correct arguments for generateGradients

lavrinovu4, 2016-05-12:
Hello all,

I am trying to generate gradients for model using command:

```Matlab
p=mymodel(); %with size(x,1) = 4, based on class Manipulator

generateGradients('dynamics',3,'dynamicsGradients',p,0,randn(4,1),0);
```
The problem is I waited a few hours, and it can't finish. Perhaps, I am using wrong arguments, or model was described not good.

There is output:
*Variable num_positions not symbolicisized
Variable num_velocities not symbolicisized
Variable num_position_constraints not symbolicisized
Variable num_velocity_constraints not symbolicisized
Variable position_constraints not symbolicisized
Variable position_constraint_ids not symbolicisized
Variable velocity_constraints not symbolicisized
Variable velocity_constraint_ids not symbolicisized
Variable joint_limit_min not symbolicisized
Variable joint_limit_max not symbolicisized
Variable joint_limit_constraint_id not symbolicisized
Variable only_loops not symbolicisized
Variable umin not symbolicisized
Variable umax not symbolicisized
Generating order 1 gradients...
Generating order 2 gradients...*

Can you help me what direction should I move?

RussTedrake, 2016-05-12:
generate gradients uses matlab's symbolic toolbox, and it's definitely possible to hand it a problem that is simply too big (disappears for hours).  the output suggests that it managed to compute the first order gradients, but got bogged down in the second order gradients.  

almost all of the routines that require gradients can also use autodiff (we call it taylorvar in the matlab code) or numerical gradients.  both will be much slower at compute time, but that might be what you'll need if your system is too complicated.

lavrinovu4, 2016-05-24:
Sorry for a delay. 
As I understood, in drake I can use a taylorApprox for generating this gradients, is it correct?
Also I found out, that the biggest number of function uses a first order derivatives. At least I will use it)

RussTedrake, 2016-05-24:
generateGradients will be the fastest solution for you.  there is also a taylorApprox method that will generate a polynomial system for you (probably not what you want) and TaylorVar which does autodifferentiation, but will be slower than running generate gradients once and writing them to a file.

closed 2016-05-25 by RussTedrake